stlink-org/stlink #1165: [STM32F407discovery]: st-flash Segmentation Fault on armbian

rusefillc, 2021-07-25:
- [ ] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [ ] Operating system an version: armbian debian Linux orangepizero 5.10.43-sunxi #21.05.6 SMP Mon Jun 21 15:07:45 UTC 2021 armv7l GNU/Linux
- [ ] **stlink tools version** and/or git commit hash: master 1.7.0-77-ga52e1bc
- [ ] stlink commandline tool name: st-flash
- [ ] Target chip (and board, if applicable): STM32F407 discovery

```
root@orangepizero(rw):/rw/stlink# uname -a
Linux orangepizero 5.10.43-sunxi #21.05.6 SMP Mon Jun 21 15:07:45 UTC 2021 armv7l GNU/Linux
root@orangepizero(rw):/rw/stlink# lsusb
Bus 008 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 007 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 003 Device 003: ID 0483:374b STMicroelectronics ST-LINK/V2.1
Bus 003 Device 002: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
root@orangepizero(rw):/rw/stlink# /rw/stlink/bin/st-flash --debug write rusefi.bin 0x08000000
st-flash 1.7.0-77-ga52e1bc
/usr/local/etc/stlink/chips: No such file or directory
2021-07-25T21:38:07 DEBUG common.c: *** looking up stlink version
2021-07-25T21:38:07 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2021-07-25T21:38:07 DEBUG common.c: stlink pid     = 0x374b
2021-07-25T21:38:07 DEBUG common.c: stlink version = 0x2
2021-07-25T21:38:07 DEBUG common.c: jtag version   = 0x19
2021-07-25T21:38:07 DEBUG common.c: swim version   = 0xe
2021-07-25T21:38:07 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-07-25T21:38:07 DEBUG usb.c: JTAG/SWD freq set to 0
2021-07-25T21:38:07 DEBUG common.c: stlink current mode: debug (jtag or swd)
2021-07-25T21:38:07 DEBUG common.c: Loading device parameters....
2021-07-25T21:38:07 DEBUG common.c: *** stlink_core_id ***
2021-07-25T21:38:07 DEBUG common.c: core_id = 0x2ba01477
2021-07-25T21:38:07 DEBUG common.c: *** stlink_read_debug32 0x410fc241 at 0xe000ed00
2021-07-25T21:38:07 DEBUG common.c: *** stlink_read_debug32 0x10076413 at 0xe0042000
Segmentation fault
```

rusefillc, 2021-07-26:
hopefully it would be useful:

tag 1.6.1 works on same exact armbian delice  just fine:

```

root@orangepizero(ro):/rw/rusefi# /rw/stlink_tag/bin/st-flash write rusefi.bin 0x08000000
st-flash 1.6.1
2021-07-26T01:02:25 INFO common.c: F4xx: 192 KiB SRAM, 1024 KiB flash in at least 16 KiB pages.
file rusefi.bin md5 checksum: 15df40a94fc0953a60484e8c3a5724bc, stlink checksum: 0x02f4f686
2021-07-26T01:02:25 INFO common.c: Attempting to write 469492 (0x729f4) bytes to stm32 address: 134217728 (0x8000000)
EraseFlash - Sector:0x0 Size:0x4000 2021-07-26T01:02:25 INFO common.c: Flash page at addr: 0x08000000 erased
EraseFlash - Sector:0x1 Size:0x4000 2021-07-26T01:02:26 INFO common.c: Flash page at addr: 0x08004000 erased
EraseFlash - Sector:0x2 Size:0x4000 2021-07-26T01:02:26 INFO common.c: Flash page at addr: 0x08008000 erased
EraseFlash - Sector:0x3 Size:0x4000 2021-07-26T01:02:26 INFO common.c: Flash page at addr: 0x0800c000 erased
EraseFlash - Sector:0x4 Size:0x10000 2021-07-26T01:02:27 INFO common.c: Flash page at addr: 0x08010000 erased
EraseFlash - Sector:0x5 Size:0x20000 2021-07-26T01:02:30 INFO common.c: Flash page at addr: 0x08020000 erased
EraseFlash - Sector:0x6 Size:0x20000 2021-07-26T01:02:32 INFO common.c: Flash page at addr: 0x08040000 erased
EraseFlash - Sector:0x7 Size:0x20000 2021-07-26T01:02:34 INFO common.c: Flash page at addr: 0x08060000 erased
2021-07-26T01:02:34 INFO common.c: Finished erasing 8 pages of 131072 (0x20000) bytes
2021-07-26T01:02:34 INFO common.c: Starting Flash write for F2/F4/L4
2021-07-26T01:02:34 INFO flash_loader.c: Successfully loaded flash loader in sram
enabling 32-bit flash writes
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 32768
size: 10740
2021-07-26T01:02:40 INFO common.c: Starting verification of write complete
2021-07-26T01:02:44 INFO common.c: Flash written and verified! jolly good!
```

Nightwalker-87, 2021-07-28:
Please test again with the proposed fix in the linked PR.

closed 2021-08-01 by Nightwalker-87