stlink-org/stlink #1174: [feature] Ability to flash F7 devices when in dual-bank mode

DaveBerkeley, 2021-08-06:
I'm working on a project using the STM32F779. We want to be able to use a bootloader to switch between flash banks bank_1 and bank_2.

I can use the latest st-flash code to write the opt_cr option bytes to set NDBANK and NDBOOT to 0. This places the device in dual-bank mode and dual-boot mode.

st-flash --area=optcr write 0x0fffaafc

Setting option_boot_add can then cause the bootloader to select code in either bank_0 (0x200) or bank_1 (0x2040).

st-flash --reset --area=option_boot_add write 0x20002000

I can't see a way to flash the part when it is in dual-bank mode. The sector erase mapping depends on the mode - the state of both of these option bits.

I've written some code to allow flashing in dual-bank mode, in a fork here, https://github.com/DaveBerkeley/stlink on branch f7-dual-bank-flash. This gets me going for my project (it can only flash bank_1, not bank_2).

st-flash --dual-bank --reset write firmware.bin 0x08000000

Am I missing something, or is this functionality missing? I'd like to contribute my changes if they are useful, but I don't know if the changes would fit in well with the existing project.

See also Application note AN4826 "STM32F7 Series Flash memory dual bank mode"

Nightwalker-87, 2021-08-15:
@DaveBerkeley Does your feature branch fully address this topic? If your approach is satisfiable, it would be great if you could share it with opening a pull request. ;-)

DaveBerkeley, 2021-08-15:
It is a bit of a hack. I added a --dual-bank switch on the command line (but not to --help) and use a table to look up the sectors to erase. I've only tested it on stm32f797 and stm32f767 parts that I have available. I have not look ed at other stm32f7 parts.
Ideally it should read the option bytes to determine the dual-bank settings, but I don't know how to do this. 
I could do this with guidance I think.

Nightwalker-87, 2021-08-15:
Ok, so are there any volunteers around? (I can't help with the F7 series at all nor do I have appropriate HW around.)

DaveBerkeley, 2021-08-15:
I've removed the --dual-bank option and added code to read the optcr reg for stm32f76x devices.

Now in https://github.com/DaveBerkeley/stlink/tree/f7_integrate branch. 

I have tested it on a Nucleo F767 board. Will try and test on my F779 target next week.

Crest, 2021-11-23:
I have access to a NUCLEO-H753 board as well as a few H750 boards (which have just a single 128kiB flash block). Should this code also work on dual banked H7xx chips or at least detect them as unsupported? If there is a way to disable dual banked mode afterward I'm could to perform the positive (and negative) testing on my H7xx boards.

antoinefaure, 2021-12-30:
Any update on this ?
I've successfully flashed the 2nd bank of my F7 with this, that's promising !

Nightwalker-87, 2023-11-27:
@Crest I've just started to review this issue. Are you still available with your hardware to take a look at the other H7-realted issues? We likely need some testing there. I'll report back on this topic soon.

Nightwalker-87, 2023-11-27:
@DaveBerkeley I've just committed a slight modification of your tested solution, taking into account recent refactoring in the codebase. Please re-test this on our `testing` branch and report back.

DaveBerkeley, 2023-11-28:
I no longer have access to the systems I was using when I opened the issue, so I can't test on a target system. It was for a commercial client that I'm no longer working for. I can pass the details on to them though. Sorry.

commit ba335a47 (2023-12-23): STM32F76xxx: Added flashing in dual bank mode

(Closes #1174)
closed 2023-12-23 by Nightwalker-87