stlink-org/stlink #1184: [STM32H743ZIT6U]: st-info --chipid retuns 0x00, works with STMCubeProgrammer

KevinLeif, 2021-09-02:
- [ x ] Programmer/board type:   
  iManufacturer           1 STMicroelectronics
  iProduct                2 STLINK-V3
  iSerial                 3 002100423438511734313939
- [ x ] Operating system an version: Ubuntu 20.04.3 LTS
- [ x ] **stlink tools version** and/or git commit hash: st-flash 1.6.1-96-gbf41f14 and st-flash 1.7.0
- [ x ] stlink commandline tool name: st-info, st-flash
- [ x ] Target chip (and board, if applicable): STM32H743ZIT6U - Nucleo-H743ZI2

Commandline output:
I get a error while flashing a binary.

```
/usr/bin/st-flash --reset --format=binary write foo.bin 0x8000000 <

st-flash 1.7.0
2021-09-02T07:49:24 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
2021-09-02T07:49:24 WARN common.c: Invalid flash type, please check device declaration
Failed to connect to target
The terminal process "bash '-c', '/usr/bin/st-flash --reset --format=binary write foo.bin 0x8000000'" failed to launch (exit code: 255).
```
Same error with the st-flash version 1.6.1. If I'm right the flash type is determined via the chip id. 
st-info 1.6.1
```
st-info --probe
Found 1 stlink programmers
 serial:     303032313030343233343338353131373334333133393339
 hla-serial: "\x30\x30\x32\x31\x30\x30\x34\x32\x33\x34\x33\x38\x35\x31\x31\x37\x33\x34\x33\x31\x33\x39\x33\x39"
 flash:      0 (pagesize: 0)
 sram:       0
 chipid:     0x0000
 descr:      unknown device

```
st-info 1.7.0
```
/usr/bin/st-info --probe
Found 1 stlink programmers
  version:    V0
  serial:     002100423438511734313939
  flash:      0 (pagesize: 0)
  sram:       24
  chipid:     0x0000
  descr:      unknown device

```

If I run the STM32CubeProgrammer instead I can connect and flash without any issues.
Stlink Firmwareversion: V3J5M2 (updated as suggested in contributing.md)

Device: STM32H7xx
Type: MCU
Device ID: 0x450
Rev: V
Flash Size: 2 MB
CPU: Cortex M7
Bootloaderversion: 0x90

Therefore I guess the hardware is fine. I've tried another board H7 board as well, same issues. 
I've checked the procedure with my hardware on a colleagues setup and everything worked fine. (1.6.1 and 1.7.0, Ubuntu20 as well)
The chip id is returned as expected. Flashing works as expected. Do I miss something? Is there additional driver or some setting I've forgot?

Nightwalker-87, 2022-01-03:
I believe the `bootrom_size` is wrong for this chip_id file anyway...
I'll check the related reference manual to verify the settings.

lucasdietrich, 2022-02-13:
I'm facing the same issue at the difference that I'm using STLink/V2-onboard. Moreover I managed to have both commands to work in the past with the version of the tool stlink-tools v1.6.0.

It stopped working and I never managed to reprogram the board using "stlink-tools" on Linux, I had to got back to on windows and program with the STM32CubeIDE/programmer.
 
I have a board `stm32f429zi (NUCLEO-f429zi)` for which all is working perfectly and constantly fine.

- [x] Programmer/board type: STLink/V2-onboard (V2.J39.M27)
- [X] Operating system an version: Ubuntu 20.04.1
- [X] tested v1.6.0 and /v1.7.0
- [X] `st-info` / `st-flash`
- [X] Target chip: STM32L011K4 (NUCLEO-L011K4)

Commandline output:
I'm getting invalid probe information with `st-info --probe` :

Version v1.7.0
```
Found 1 stlink programmers
  version:    V0
  serial:     FF555557838667143150
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0xffffffff
```

Version v1.6.0
```
lucas@laptop-dev:~$ st-info --probe
Found 1 stlink programmers
 serial: 303636444646353535353537383338363637313433313530
openocd: "\x30\x36\x36\x44\x46\x46\x35\x35\x35\x35\x35\x37\x38\x33\x38\x36\x36\x37\x31\x34\x33\x31\x35\x30"
  flash: 0 (pagesize: 0)
   sram: 0
 chipid: 0x0000
  descr: unknown device
```

And then I can't erase : using `st-flash erase`:
```
st-flash 1.7.0
2022-02-13T18:08:15 INFO common.c: Loading device parameters....
2022-02-13T18:08:15 WARN common.c: Invalid flash type, please check device declaration
Failed to connect to target
```

Expected/description:

When It was working in the past (st-flash also worked at this moment).
```
Found 1 stlink programmers
 serial: 303636444646353535353537383338363637313433313530
openocd: "\x30\x36\x36\x44\x46\x46\x35\x35\x35\x35\x35\x37\x38\x33\x38\x36\x36\x37\x31\x34\x33\x31\x35\x30"
  flash: 16384 (pagesize: 128)
   sram: 8192
 chipid: 0x0457
  descr: L011 device
```

Hope it helps

Nightwalker-87, 2022-02-16:
@lucasdietrich Thanks for the input. However the comparison with older versions does not help here due to larger refactoring work and restructuring in the library recently. One would clearly need to review the current state in the `develop` branch.

Nightwalker-87, 2022-12-28:
@lucasdietrich Sorry for the long delay. I finally found some time to review some issues.
Do you still have the setup / toolset to give this another try? If so, one should solely focus on Release v1.7.0.

Nightwalker-87, 2023-01-02:
Support for STM32H7 devices was not present before Release v1.7.0.
This feature was added to the `develop` branch in commit 1e2092173aae4ce0c256ff345cf647ec868f5435.

@lucasdietrich Your attempt with the L0 device leads to the same result as with the H7, as it is running through the same routine in the code, but must have a different cause. Need to investigate further which then may reduce efforts for bisecting.

Nightwalker-87, 2023-01-02:
@lucasdietrich Looking closer at this, I believe you should try to reproduce your case with Release v1.7.0.
Earlier releases do not compile any more with recent toolchains, so this is clearly not an option for any debugging attempts on the recent dev environment I use to maintain the codebase.

For users with a STM32L011K4 and/or a STM32H743ZIT6U device and an STLink/v2 programmer (we may extend the testing to the STLink/v3 in a second step):

Please proceed as follows:
1) Checkout Release v1.7.0 via git
2) Update your programmers to the latest firmware available from ST-Microelectronics:
[ST-LINK, ST-LINK/V2, ST-LINK/V2-1, STLINK-V3 boards firmware upgrade](https://www.st.com/en/development-tools/stsw-link007.html)
2) Run the following command sequence in the working directory:
```
sudo make clean
sudo make install && sudo make uninstall && sudo make clean
sudo make install
st-info --probe
```

... and report back the output you receive.

Nightwalker-87, 2023-01-02:
@fdelbos I just noticed in #1253 that you have a L011K4 Nucleo. Can you help testing?

fdelbos, 2023-01-02:
@Nightwalker-87 sorry about that, i m remote working for a few months and wont have access to the kits until spring.

Nightwalker-87, 2023-01-02:
@fdelbos Thanks for the quick response. Well never mind... :neutral_face:

lucasdietrich, 2023-01-02:
Did the steps you mentionned above, I'm having the following result with a NUCLEO-l011k4 board :
```
lucas@vmdev:~/stlink$ ./build/Release/bin/st-info --probe
Failed to parse flash type or unrecognized flash type
Found 1 stlink programmers
  version:    V2J31S21
  serial:     066BFF555557838667142135
  flash:      16384 (pagesize: 128)
  sram:       8192
  chipid:     0x457
  dev-type:   STM32L0xxx_Cat_1
```

- Ubuntu 20.04.4 LTS
- ST Link firmware version: V2J39M27 (latest proposed by STCubeProgrammer)

Moreover erasing and programming the chip works for me now 👍

Nightwalker-87, 2023-01-02:
@lucasdietrich Oh, that is good news! Thanks for the feedback. :1st_place_medal:

Nightwalker-87, 2023-01-02:
@EmanueleGiacomini I've noticed in #1249 that you have similar hardware which would qualify for testing here.
Could you give it a try? It would be great and may allow to finally resolve this issue...

Nightwalker-87, 2023-01-02:
Alternatively: @ronangaillard and/or @benjinne - you may also give it a try, if you fancy.

benjinne, 2023-01-02:
I don't remember the exact issue I had, but around Oct 25th, the latest release wasn't working so I built it from the main branch and it started working

Nightwalker-87, 2023-01-02:
@benjinne Oh well, one may give it a try on the current `develop` with the same procedure as described earlier,
but I expect it to work as well. At least, it does not happen on my STM32F10x boards with STLink/v2 which are the only devices I have around for testing currently.

defyingphysics, 2023-01-08:
I have tested using a Nucleo H7 A3ZI-Q 
When using the dev branch (note ./, run from /bin in dev branch with sha: d053470): 
```
~/Desktop/stlink/bin  develop
❯ ./st-info --probe
/usr/share//stlink/chips: No such file or directory
Found 1 stlink programmers
  version:    V3J10
  serial:     004F003E3038510734333935
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x480
```
However it seems the 1.7.0 release is working just fine. ( have installed 1.7.0)
```
~/Desktop/stlink/bin  develop
❯ st-info --probe
Found 1 stlink programmers
  version:    V3J10
  serial:     004F003E3038510734333935
  flash:      2097152 (pagesize: 8192)
  sram:       131072
  chipid:     0x0480
  descr:      H7Ax/H7Bx

```

Nightwalker-87, 2023-01-08:
@defyingphysics Thank you for re-testing. Very interesting indeed...
Can you git bisect between latest `develop` and the **1.7.0 Release** (commit: b97296be8f22421b5beb21a4462bf8e515f74ab8)?
We now need to find the relevant change to fix this.

Nightwalker-87, 2023-01-08:
Further relevant interim commits to look at:
- dc5388d94c959b271d51cd98ea973cf4e8febcf0 <-- most suspicious
- 1c5446a91eff0b893b5d937bf854d040ed75109f
- 66f35d1331ff0a7df12c7692f10f805b3bd75e36

davidgiven, 2023-02-14:
I'm getting the same thing, _with_ 1.7.0 (on Debian Linux sid). I'm using a RobotDyn ST32 MINI Black Pill. STM32CubeProgrammer reports this as an STM32F101/F102/F103, device ID 0x410, Rev X, Cortex-M3. st-info --probe says this:

```
$ st-info --probe 
Found 1 stlink programmers
  version:    V2J40S7
  serial:     352605012A12354D314B4E00
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device
```

davidgiven, 2023-02-14:
...addendum to the last: after removing the two boot mode jumpers st-info worked fine... and then after reinserting the jumpers it continues to work fine.

I have no idea what happened or why, but at least it's working now.

Nightwalker-87, 2023-02-14:
Very interesting indeed... This sounds as if a temporary GND-disconnect could play a role...

davidgiven, 2023-02-26:
For reference: what _are_ the canonically correct jumper positions for flashing? I flashed a firmware which turns off the debugging ports, and now I can't make st-link work at all! This suggests that I only had it working at all previously because it was booting into the user application which _happened_ to leave the debug ports open. There should be a jumper setting for forcing it into the bootloader, right?

Aukstkalnis, 2023-02-27:
Adding --connect-under-reset helps to receive device info, but after st-info exits device is left in reset state.

Nightwalker-87, 2023-03-19:
Concluding review:

The testing results from @lucasdietrich and @defyingphysics verify that this issue is fixed.
Investigation of this issue reveals that necessary support for H7 devices made it into the `master` branch with #1099 on 11 Mar 2021 together with several other changes. By that time this was unintended and by mistake, as these topics were all still considered under development by then. Nevertheless we can safely assume that by the official release of version v1.7.0 on 25 Apr 2021 all necessary changes were complete. Apart from this we may assume that there were some leftovers from previous installations causing the original issue to appear. Here we recommend to do a manual clean-up of remnant files and to start over again with a clean install.

In this context we can mark this issue as resolved by **Release v1.7.0** and even more precise by #1052.

closed 2023-03-19 by Nightwalker-87