stlink-org/stlink #1189: [STM32F103]: undefined symbol: init_chipids

MrZloHex, 2021-09-16:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [X] Programmer/board type: [STLINK V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [X] Operating system an version: [Fedora 34] (e.g Linux, macOS, Windows)
- [X] **stlink tools version** and/or git commit hash: [v1.7.0-105-gdb8f789] (e.g v1.6.1/git-d0416149)
- [X] stlink commandline tool name: [st-util] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [X] Target chip (and board, if applicable): [STM32F103] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

I'm trying troubleshoot and debug some code on board with `st-util` and `arm-none-eabi-gdb`, but i can't startup server with `st-util`.

Commandline output:

```
❯ st-util
st-util 1.7.0-105-gdb8f789
st-util: symbol lookup error: st-util: undefined symbol: init_chipids
```

Expected/description:

I expected startup of gdb server on port 4242.

slyshykO, 2021-09-16:
Hi, please show us an output of ldd for st-util.

MrZloHex, 2021-09-16:
Hello,
```
❯ ldd /usr/local/bin/st-util
	linux-vdso.so.1 (0x00007ffeb0fe8000)
	libstlink.so.1 => /lib64/libstlink.so.1 (0x00007f06f9786000)
	libusb-1.0.so.0 => /lib64/libusb-1.0.so.0 (0x00007f06f9769000)
	libc.so.6 => /lib64/libc.so.6 (0x00007f06f959a000)
	libudev.so.1 => /lib64/libudev.so.1 (0x00007f06f956e000)
	libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f06f954d000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f06f97bf000)
	librt.so.1 => /lib64/librt.so.1 (0x00007f06f9542000)
	libselinux.so.1 => /lib64/libselinux.so.1 (0x00007f06f9514000)
	libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f06f94f9000)
	libpcre2-8.so.0 => /lib64/libpcre2-8.so.0 (0x00007f06f9462000)
	libdl.so.2 => /lib64/libdl.so.2 (0x00007f06f945b000)
```

slyshykO, 2021-09-16:
Try to run LD_LIBRARY_PATH=/usr/local/lib /usr/local/bin/st-util

MrZloHex, 2021-09-16:
```
❯ LD_LIBRARY_PATH=/usr/local/lib /usr/local/bin/st-util
st-util 1.7.0-105-gdb8f789
/usr/local/bin/st-util: symbol lookup error: /usr/local/bin/st-util: undefined symbol: init_chipids
```

slyshykO, 2021-09-16:
@MrZloHex 
The issue is that the newly built st-util loaded with the old libstlink is probably installed through the system packet manager.

Do you remember or can find where is installed *libstlink.so.1.7.0 from you build?

MrZloHex, 2021-09-16:
I installed `stlink tools` as `sudo dnf install stlink` and you think, that it installed old version of library?

Should I find the path to _libstlink.so.1.7.0_?

slyshykO, 2021-09-16:
You should find out where is installed `libstlink.so*` which build with `st-util 1.7.0-105-gdb8f789`.
Then force the system to run `st-util 1.7.0-105-gdb8f789` with proper `libstlink.so*`.
Sorry, I don't know how to do it on fedora.

MrZloHex, 2021-09-16:
```
❯ sudo find | grep libstlink
[sudo] password for zs: 
find: ‘/run/user/1000/gvfs’: Permission denied
/usr/lib64/libstlink.so
/usr/lib64/libstlink.so.1
/usr/lib64/libstlink.so.1.7.0
```

But how to understand which of those is using for `st-util`?

slyshykO, 2021-09-16:
@MrZloHex, where you get `st-util 1.7.0-105-gdb8f789`?

MrZloHex, 2021-09-17:
@slyshykO 
Like i just execute `st-util`:
```sh
❯ /usr/local/bin/st-util
st-util 1.7.0-105-gdb8f789
/usr/local/bin/st-util: symbol lookup error: /usr/local/bin/st-util: undefined symbol: init_chipids
❯ /usr/local/bin/st-util --version
v1.7.0-105-gdb8f789
```

Nightwalker-87, 2021-09-20:
@MrZloHex Yes, where does that revision come from?

MrZloHex, 2021-09-21:
from `dnf` repo, i installed it like any prog/lib with dnf

mariobaldini, 2021-11-02:
Syncing installed lib with the st-link bin solved the issue in my case:

Purged both the lib and installed bin: `sudo apt remove --purge libstlink1 stlink`
Reinstalled the bin again: `sudo dpkg -i stlink_1.7.0-1_amd64.deb`

```
ldd /usr/bin/st-flash
    linux-vdso.so.1 (0x00007fff976f5000)
    libstlink.so.1 => /lib/libstlink.so.1 (0x00007fe39a984000)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe39a75c000)
    libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fe39a73f000)
    /lib64/ld-linux-x86-64.so.2 (0x00007fe39a9c0000)
    libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fe39a716000)
    libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe39a711000)
```
libstlink.so.1 was previously = 0x00007f88c8e4c000

Probably it's also the same solution for Fedora (just switching from apt to dnf). 

Apparently when uninstalling `stlink` pkg it doesn't remove `libstlink1`, so when it get manually installed it keeps the older lib. 

Thanks @slyshykO

Nightwalker-87, 2022-01-03:
@mariobaldini @MrZloHex Please report this to the Fedora package maintainer downstream.
We can't take action here, as we don't split-up the _stlink_ package into several parts and only distribute a single package instead.

closed 2022-01-03 by Nightwalker-87