stlink-org/stlink #1191: [ST-LINK /V2]: Problems running when installing from binaries

nicosquare, 2021-09-23:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue
- [x] Programmer/board type: [STLINK /V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [Linux] (e.g Linux, macOS, Windows)
- [x] **stlink tools version** and/or git commit hash: [v1.7.0] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [`st-info`, `st-flash`, `st-trace`, `st-util`] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ ] Target chip (and board, if applicable): [enter here] (e.g STM32F103C8T6 (NUCLEO-F103RB))

I want to use the latest version of st-link as I understand that it will help me solve an issue while programming an STM32 board. I followed the instructions in the compilation manual and apparently, it is installed correctly. However when I run any command of st-link it shows the following error:
 
Commandline output:

```
st-info --version
st-info: symbol lookup error: st-info: undefined symbol: arg_parse_freq
```

Expected/description:

`I'm expecting to get the version of st-link with that simple command`

slyshykO, 2021-09-23:
Think this issue connected with #1189

mariobaldini, 2021-11-02:
Syncing installed lib with the st-link bin solved the issue in my case:

Purged both the lib and installed bin: `sudo apt remove --purge libstlink1 stlink`
Reinstalled the bin again: `sudo dpkg -i stlink_1.7.0-1_amd64.deb`

```
ldd /usr/bin/st-flash
    linux-vdso.so.1 (0x00007fff976f5000)
    libstlink.so.1 => /lib/libstlink.so.1 (0x00007fe39a984000)
    libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fe39a75c000)
    libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fe39a73f000)
    /lib64/ld-linux-x86-64.so.2 (0x00007fe39a9c0000)
    libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fe39a716000)
    libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fe39a711000)
```
libstlink.so.1 was previously = 0x00007f88c8e4c000

Probably it's also the same solution for Fedora (just switching from apt to dnf). 

Apparently when uninstalling `stlink` pkg it doesn't remove `libstlink1`, so when it get manually installed it keeps the older lib.

Nightwalker-87, 2022-01-03:
@nicosquare Which linux distribution do you use?

nicosquare, 2022-01-03:
> @nicosquare Which linux distribution do you use?

Hi! @Nightwalker-87 I use Ubuntu 20.04

Nightwalker-87, 2022-01-03:
Ah, ok then we have the same issue as in #1189...

Please report this to the Debian/Ubuntu package maintainer downstream.
We can't take action here, as we don't split-up the stlink package into several parts and only distribute a single package instead.
To avoid this straight away I recommend to use the latest Debian package distributed from this project (local package installation), which is known to be working. Thanks for the feedback!

closed 2022-01-03 by Nightwalker-87