stlink-org/stlink #1209: [feature] GD32F190 support

rusefillc, 2022-01-02:
- [ ] Programmer/board type: STLINK /V3
- [ ] Operating system an version: Windows 11
- [ ] **stlink tools version** and/or git commit hash: 1.7.0 release
- [ ] stlink commandline tool name: st-info
- [ ] Target chip (and board, if applicable): gd32f190

I am trying
st-info.exe  --probe 

and all I get is
```
Found 1 stlink programmers
  version:    V3J7
  serial:     001600095553500F20393256
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0000
  descr:      unknown device
```

is there a way for me to enable verbose debug output if you guys have any interest in support of this interesting 5v chip with 0x1ba01477 core id?

I realize that openocd and stlink are very different things but let me attach openocd verbose output just for reference
[gd32f190_stlink-v3.txt](https://github.com/stlink-org/stlink/files/7799896/gd32f190_stlink-v3.txt)

Nightwalker-87, 2022-01-02:
No, we don't support clone devices and they often cause problems regarding identification based on chip parameters.
As settings do interfere with original chips we won't mix that up again. We have learned this lesson in the project history already.

rusefillc, 2022-01-02:
that's fair I guess. if GD32 wanted to be taken care seriously they should not have used same core id.

Nightwalker-87, 2022-01-02:
True. I recommend you to look for an original development board from ST-Micro for your application based on your specific needs. Some models are available for less than 20€, but may save time spent on getting it to work or fixing issues with spare documentation available or may even prevent to buy several cheap clones in order to finally get to a working device. I've seen this before with a Bluepill board and some Chinese STLINK/V2 clone programmers. Take it as a good advice to prevent frustration without the intention to promote anything. :wink:

rusefillc, 2022-01-03:
I assume you are aware of the chip shortage right? :) while some dev boards are still available have you seen how official F407 discovery is not planning to be available for the next 12 months?

I am well past dev board phase. I am having troubles fabricating my devices like https://github.com/rusefi/rusefi-hardware/tree/main/wideband-F103

Nightwalker-87, 2022-01-03:
Of course I am, but this should turn out to be the same situation for everyone.
However I am closing this issue now as it appears to be resolved.

closed 2022-01-03 by Nightwalker-87