stlink-org/stlink #1211: usb.c refactoring request: can we remove some unused code?

hydroconstructor, 2022-01-06:
In usb.c there are next lines:

```
stlink_t *stlink_open_usb(...) {
line 1144:
    // TODO: Reading a environment variable in a usb open function is not very nice, this should
    // be refactored and moved into the CLI tools, and instead of giving USB_BUS:USB_ADDR a real
    // stlink serial string should be passed to this function. Probably people are using this
    // but this is very odd because as programmer can change to multiple busses and it is better
    // to detect them based on serial.
    char *device = getenv("STLINK_DEVICE");

```
Here supposed that there is an environment variable in user system with name "STLINK_DEVICE" and value "Device_bus:Device_Address". In case of presence such variable in user system function gets device bus and device address from USB devices list and compare them with ones in environment variable. If do not match then proceed with next USB device from list.

There is no such variable in my system. I can not find any place in code which set this variable. Function stlink_open_usb gets serial of ST-LINK in 3rd parameter and uses it to find right USB device. So may it be useful to completely remove all code which works with getenv function and variables device, devBus and devAddr?

hydroconstructor, 2022-01-06:
Pushed new commit #1210 pull request

Nightwalker-87, 2022-01-06:
In general it is favourable to do so if there is no remaining use within the codebase.
However this must be assured for all platforms.

Please note that no further refactoring should take place until the chip-id reading has been finally rewritten.
All old bindings need to be replaced and switched to new structure.

PR #1210 (2022-01-23): Fix for libusb_kernel_driver_active & error handling for st.st_size
According to libusb API documentation, function **_libusb_kernel_driver_active_** is not available on Windows. So I think it will be useful not to compile it on Windows platform.

(Closes #1211) (Closes #1214)
closed 2022-01-23 by Nightwalker-87