stlink-org/stlink #1212: [refactoring] Transition to chipid-specific device-files

Nightwalker-87, 2022-01-06:
The following tasks should be achieved to improve the codebase of the toolset:

- modularize the codebase
- reduce cross-dependencies between headers and source files
- separate device-specific settings from source code (#1129)
- semantic versioning (#1075)
- use of application binary interfaces (ABI) to improve compatibility (#1075)

Nightwalker-87, 2022-01-06:
I suggest the following workflow for identifying a MCU:

1. Read .chip file (core-id / chip-id)
2. Check to verify correctness of memory addresses
3. Print identified MCU to console output / error message on failure
3. Translate verified values to human-readable defines for further use throughout the codebase
4. TBC...

Nightwalker-87, 2022-01-06:
`stlink_stm32_chipids` should actually go to `stm32.h` as it is STM32 device specific.
According to the previous suggestion the related `enum` could be replaced with an `if`-structure like:
```
if $(hex-address_1) {
   #define $(name_of_define_1) $(hex-address_1)
} else if ($(hex-address_2) {
   #define $(name_of_define_2) $(hex-address_2)
} else if ...
```

Nightwalker-87, 2022-01-09:
@Ant-ON I have just pushed a new commit to branch `chip_cleanup` with some further preparations.
How can we properly switch the implementation to read parametres from the new chipid files? Can/should we achieve this stepwise?

Nightwalker-87, 2022-01-09:
Based on the original intention to solely complete the separation of chip-specific parametres out of the source code and into device-specific files led to some more general clean-up and restructuring. Nevertheless work is not complete of course...
I believe that the source code and header files `stm32.h`, `chipid.c` and `chipid.h` are on a good way to reach structure and content as we may expect them to be and to serve as.

For now I am going to make a halt here to give everybody a chance to follow the changes which appear quite huge already.
Please let me know if you deserve a PR to be opened before continuing with other topics.

Nightwalker-87, 2022-01-09:
... and BTW: When does

```
#ifdef __cplusplus
    extern "C" {
#endif

[...]

#ifdef __cplusplus
}
#endif
```

exactly make sense?

slyshykO, 2022-01-10:
> ... and BTW: When does
> 
> ```
> #ifdef __cplusplus
>     extern "C" {
> #endif
> 
> [...]
> 
> #ifdef __cplusplus
> }
> #endif
> ```
> 
> exactly make sense?

It needs for whom who uses our includes in C++ code.

Nightwalker-87, 2022-01-10:
Tell me a story, I got that far, but the question is: Where is it necessary from the technical perspective. It is not in general, because otherwise it would be defined in every source code file, which isn't the case and compilation and execution in C++ is known to be working anyway.

hydroconstructor, 2022-01-10:
What is "modularisation of common.c"? Divide it to several files?
What is cross-dependencies between headers and source files? Rewrite sources to reduce number of included headers for each source?

slyshykO, 2022-01-10:
> Where is it necessary from the technical perspective?

It is necessary when someone uses libstlink in a custom C++ application because C++ mangle functions names by default and without `extern "C"` declaration linker didn't find proper symbols.

Also, this extern guard is not needed in all sources. It is required only in the main header, for example, in **stlink.h** in our case.

Nightwalker-87, 2022-01-10:
@hydroconstructor `common.c` is basically an bunch of stuff, partially hardcoded and/or duplicated from implementations that were written later. Basically spoken it's the historical root of the whole project and holds many remnants from earlier times. We want that sorted, separated and integrated into the source code structure that was introduced and extended by many valuable contributions in the more recent years. "Modular" basically means that different parts of the code that handle different operations or functionality should not mix up and should only have cross dependencies that are either really necessary on the technical level or implemented in such a way that changes in one module or specific part do not imply several other necessary changes in other parts of the code.

Nightwalker-87, 2022-01-11:
Please test the branch `chipid_cleanup` with various boards and report any issues.

gszy, 2022-01-16:
:large_orange_diamond: As discussed with @Nightwalker-87, I’m adding another cleanup-related idea, originally described at <https://github.com/stlink-org/stlink/commit/c854df5edd1285c1016e6181bcd54acf6f2e0bcd#commitcomment-63941871>.

`enum stm32_flash_type` describes the different flash types that STM32 MCUs have. The enumeration constants have different, manually assigned, integer values, beginning with special `STM32_FLASH_TYPE_UNKNOWN = 0` and ending with special `STM32_FLASH_TYPE_UNDEFINED = 12`. I’ve suggested removing the manual assignments, as the C compiler would automatically assign precisely the same values, but I :zombie: forgot the—somewhat arbitrarily chosen—values have actual practical meaning: they are used in the chip description files as the values of the `flash_type` property.

I suggest the following:

- the chip description `flash_type` property should be a human-readable string describing the flash type, for example `L4_L4P` or `G0`;
- `enum stm32_flash_type` constant values should be automatically assigned, with `STM32_FLASH_TYPE_UNKNOWN` being the first and `STM32_FLASH_TYPE_UNDEFINED` being the last;
- whichever stlink part parses chip files, should convert the string descriptions to the enum values.

Nightwalker-87, 2022-01-18:
@gszy I've replaced the list of defines with an enum holding the flash types in human readable form and added the individual types to the .chip files, replacing the integer values there. However this implies that the following excerpt in `void process_chipfile(char *fname) { }` in `chipid.c` needs to be rewritten to use the enum and convert the respective types from `enum` to `int`.

```
[...]
        } else if (strcmp (word, "flash_type") == 0) {
            if (sscanf(value, "%i", (int *)&ts->flash_type) < 1) {
                fprintf(stderr, "Failed to parse flash type\n");
            } else if ((ts->flash_type < STM32_FLASH_TYPE_UNKNOWN) || (ts->flash_type >= STM32_FLASH_TYPE_UNDEFINED)) {
                fprintf(stderr, "Unrecognized flash type\n");
            }
[...]
```

I'm unsure yet on the most reasonable approach. So far this unfinished attempt is only a local version for testing purposes.

gszy, 2022-01-20:
The simplest solution would be a bunch of this kind of comparisons:

```c
if (strcmp(value, "G0") == 0) {
    &ts->flash_type = STM32_FLASH_TYPE_G0;
}
```

It may be a part of `process_chipfile()`, or extracted to a separate function.

Alternatively we could have a const array somewhere that would map the enum values to the string values, something like:

```c
const char *x[] = {
    [STM32_FLASH_TYPE_G0] = "G0",
};
```

then loop through the array when parsing a flash type.

Nightwalker-87, 2022-01-20:
I'd prefer the first attempt to make it work. Anything seems better than the `#defines` used currently. A first implementation is found in  commit 8d96e769f32d1c85be913f77e6aab4b1d84b80b1. Please comment in #1216 respectively.

PR #1216 (2022-01-28): [refactoring] General maintenance for code structure
List of changes and improvements:

- Revised device-specific parametres in the .chip files
- Aligned naming for .chip files and defines
- Unified formatting for .chip files
- Removed function `dump_a_chip` and old device database from `common.c`
- Moved-out STM32-specific defines to `stm32.h`
- Updated MCU core-ids
- Removed additional `extern "C"` linkage specs

(Closes #903) (Closes #1090) (Closes #1199) (Closes #1212)
closed 2022-01-28 by Nightwalker-87