stlink-org/stlink #1215: Nucleo H755ZI-Q: Unable to use different Access Points

TheEiken, 2022-01-14:
Hallo,

I'm using dual core processors of the H7 family, here the H755 of the named Nucleo board.
In the option bytes I've set the BCM7 flag to disabled. so the CM7 core does not start. After this, for debugging/flashing the access port 0 of the CM7 is inactive, so another access port must be selected (1-3 in this case).
It seems that currently the stlink tools choose only the default access port 0. I did not found any possibility to change the access port. Without selecting another access port it is not possible to debug or flash the controller, if the default access port is not accessable.

The used stlink version was 1.7.0 and the current the develop version (g 7cc1fda) on Ubuntu 20.04LTS.

This is my first use of github, so if you miss something, let me know.

Regards
Andreas Eiken

Output of st-flash:
tools/stlink/st-flash --debug --reset write /data/develop/projects/kdr_bl_devel/990-Source/bin/debug/kdr_cm4_bl_0_0_1_48076_HW_0_01_d.fwf 0x8100000
st-flash 1.7.0
2022-01-14T11:50:35 DEBUG common.c: *** looking up stlink version
2022-01-14T11:50:35 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2022-01-14T11:50:35 DEBUG common.c: stlink pid     = 0x374e
2022-01-14T11:50:35 DEBUG common.c: stlink version = 0x3
2022-01-14T11:50:35 DEBUG common.c: jtag version   = 0x8
2022-01-14T11:50:35 DEBUG common.c: swim version   = 0x0
2022-01-14T11:50:35 DEBUG common.c: stlink current mode: mass
2022-01-14T11:50:35 DEBUG usb.c: JTAG/SWD freq set to 0
2022-01-14T11:50:35 DEBUG common.c: stlink current mode: mass
2022-01-14T11:50:35 DEBUG common.c: *** stlink_enter_swd_mode ***
2022-01-14T11:50:35 DEBUG common.c: Loading device parameters....
2022-01-14T11:50:35 DEBUG common.c: *** stlink_core_id ***
2022-01-14T11:50:35 DEBUG common.c: core_id = 0x6ba02477
2022-01-14T11:50:35 DEBUG usb.c: READDEBUGREG error (0x05)
2022-01-14T11:50:35 ERROR common.c: Can not connect to target. Please use 'connect under reset' and try again
Failed to connect to target
2022-01-14T11:50:35 DEBUG common.c: *** stlink_exit_debug_mode ***
2022-01-14T11:50:35 DEBUG common.c: *** stlink_close ***

c-grant, 2022-01-27:
This is related to issue #1160

Nightwalker-87, 2023-11-21:
Closing this feature request due to inactivity and obvious lack of general interest.

closed 2023-11-21 by Nightwalker-87