stlink-org/stlink #1225: [stm32l053c8t6]: Cannot flash with Release version but can do with Debug

mynameisdaniil, 2022-02-14:
Current develop version of st-link tools cannot flash subj MCU when I'm using locally built release version on macos@arm64, though it works fine when I'm using debug build. Timing issues?
Furthermore, nor debug nor release versions work on linux@amd64 but v1.6 provided in ubuntu repo works just fine.
Almost went into binge debug mode but thought somebody might know answer from the top of their head.

- [x] Programmer/board type: V2-onboard (updated to the latest firmware available on the date of this posting)
- [x] Operating system an version: macOS 12.2.1@arm64/ubuntu 20.04.3 (5.13.0-27-generic)@amd64
- [x] **stlink tools version** and/or git commit hash: develop (468b1d2daa853b975c33ab69876c486734f2c6a7)
- [x] stlink commandline tool name: `./build/Debug/bin/st-flash --reset write ./firmware.bin 0x08000000`
- [x] Target chip (and board, if applicable): stm32l053c8t6(stm32l0-discovery)

<details>
  <summary>macos/release:</summary>
  
```
$ ./build/Release/bin/st-flash --reset write ./firmware.bin 0x08000000
st-flash 1.7.0-184-g468b1d2-dirty
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32L0xxx_Cat_3
# Reference Manual: RM0451           // also RM0367 & RM0377
#
chip_id 0x417
flash_type 8
flash_size_reg 0x1ff8007c
flash_pagesize 0x80
sram_size 0x2000
bootrom_base 0x1ff00000
bootrom_size 0x1000
option_base 0x1ff80000
option_size 0x20
flags 0

2022-02-14T13:57:00 INFO common.c: STM32L0xxx_Cat_3: 8 KiB SRAM, 64 KiB flash in at least 128 byte pages.
file ./firmware.bin md5 checksum: fbe4a7bb1c1f4fd6288d7a8721fb79, stlink checksum: 0x000fa383
2022-02-14T13:57:00 INFO common_flash.c: Attempting to write 10940 (0x2abc) bytes to stm32 address: 134217728 (0x8000000)
-> Flash page at 0x8000000 erased (size: 0x80)
-> Flash page at 0x8000080 erased (size: 0x80)
-> Flash page at 0x8000100 erased (size: 0x80)
-> Flash page at 0x8000180 erased (size: 0x80)
-> Flash page at 0x8000200 erased (size: 0x80)
-> Flash page at 0x8000280 erased (size: 0x80)
-> Flash page at 0x8000300 erased (size: 0x80)
-> Flash page at 0x8000380 erased (size: 0x80)
-> Flash page at 0x8000400 erased (size: 0x80)
-> Flash page at 0x8000480 erased (size: 0x80)
-> Flash page at 0x8000500 erased (size: 0x80)
-> Flash page at 0x8000580 erased (size: 0x80)
-> Flash page at 0x8000600 erased (size: 0x80)
-> Flash page at 0x8000680 erased (size: 0x80)
-> Flash page at 0x8000700 erased (size: 0x80)
-> Flash page at 0x8000780 erased (size: 0x80)
-> Flash page at 0x8000800 erased (size: 0x80)
-> Flash page at 0x8000880 erased (size: 0x80)
-> Flash page at 0x8000900 erased (size: 0x80)
-> Flash page at 0x8000980 erased (size: 0x80)
-> Flash page at 0x8000a00 erased (size: 0x80)
-> Flash page at 0x8000a80 erased (size: 0x80)
-> Flash page at 0x8000b00 erased (size: 0x80)
-> Flash page at 0x8000b80 erased (size: 0x80)
-> Flash page at 0x8000c00 erased (size: 0x80)
-> Flash page at 0x8000c80 erased (size: 0x80)
-> Flash page at 0x8000d00 erased (size: 0x80)
-> Flash page at 0x8000d80 erased (size: 0x80)
-> Flash page at 0x8000e00 erased (size: 0x80)
-> Flash page at 0x8000e80 erased (size: 0x80)
-> Flash page at 0x8000f00 erased (size: 0x80)
-> Flash page at 0x8000f80 erased (size: 0x80)
-> Flash page at 0x8001000 erased (size: 0x80)
[... 426 lines cut ...]
-> Flash page at 0x8002680 erased (size: 0x80)
-> Flash page at 0x8002700 erased (size: 0x80)
-> Flash page at 0x8002780 erased (size: 0x80)
-> Flash page at 0x8002800 erased (size: 0x80)
-> Flash page at 0x8002880 erased (size: 0x80)
-> Flash page at 0x8002900 erased (size: 0x80)
-> Flash page at 0x8002980 erased (size: 0x80)

2022-02-14T14:00:34 INFO flashloader.c: Starting Flash write for L0
2022-02-14T14:00:34 INFO flash_loader.c: Successfully loaded flash loader in sram
2022-02-14T14:00:34 INFO flash_loader.c: Clear DFSR
2022-02-14T14:00:34 INFO common.c: Go to Thumb mode
2022-02-14T14:00:35 ERROR flash_loader.c: Flash loader run error
2022-02-14T14:00:35 WARN flash_loader.c: Loader state: R2 0x8 R15 0x7FDF
2022-02-14T14:00:35 WARN flash_loader.c: MCU state: DHCSR 0x1080009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2022-02-14T14:00:35 WARN flashloader.c: Failed to use flash loader, fallback to soft write
167/167 halfpages written
2022-02-14T14:00:35 ERROR common_flash.c: Invalid flash address
stlink_fwrite_flash() == -1
```
</details>

Expected/description:

I expect it to work in all cases.

mynameisdaniil, 2022-02-18:
I'd add `os/linux` too, since it doesn't work on Linux either. It's just easier to debug on macos since it works in case of debug build.

Nightwalker-87, 2022-12-29:
@gszy Do you have an idea on this one? There appears to be a timeout leading to the first error:

```
    if (timeout) {
        ELOG("Flash loader run error\n");
        goto error;
    }
```

`flashloader.c: Failed to use flash loader, fallback to soft write` derives from the old implementation as far as I am concerned. `flash_loader.c` has been written by @chenguokai and is a new approach from scratch as to my knowledge. Maybe there is a conflict between both upon being called ...

Nightwalker-87, 2022-12-29:
@mynameisdaniil When looking at your outputs, we find that there is erroneous behaviour in all four of them, even though you claim the "macOS Debug" case to be working - the flash loader isn't doing so.

Nightwalker-87, 2023-01-07:
@Ant-ON This is another strange topic where I'm unsure where to look at really. Is it possibly timeout-related only or is there an improper way of implementation when trying to select the flash loader? I couldn't make it out yet.

mynameisdaniil, 2023-02-04:
Is there any way I can help you?

Nightwalker-87, 2023-02-04:
Thanks for the offering @mynameisdaniil. The current problem I have here is a missing clue where to look at specifically. The people most recently involved in or related topics around this implementation have been triggered already. We're still waiting for any response or anyone else familiar with the topic. I'd be looking forward to resolve this issue as well...

Nightwalker-87, 2023-03-19:
Related to #1253 and #1289.

Nightwalker-87, 2023-04-30:
Duplicate of #681.

Nightwalker-87, 2023-06-17:
@mynameisdaniil Please continue to follow the main thread in #681.
I've just reviewed and summarized the relevant facts there.

mynameisdaniil, 2023-06-17:
@Nightwalker-87 will do. Thanks for the heads-up.

Ant-ON, 2023-06-18:
@Nightwalker-87 I think this might be another problem with halfpage write mode. Need to check the changes suggested in #1203. Yes, this changes is to completely remove halpage mode, but it is better to be slower than with errors...

Nightwalker-87, 2023-06-18:
@Ant-ON We should address everything in #681, as the derived result is at least very similar if not the same. It does not matter should there be several reasons of origin. It is difficult to track and discuss an issue in several threads. I've added the relevant info from other related tickets there as well.

Ant-ON, 2023-06-18:
@Nightwalker-87 
For reasons not clear to me (most likely due to the state of MCU) does not work writing by half page. Without hardware this problem is hard to find. I think the best solution at the moment is to disable this code.

closed 2023-09-01 by Nightwalker-87