stlink-org/stlink #1228: [STM32 device name]: STM32F446 wrong chip definition

UserX404, 2022-02-23:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [X] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [ ] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [ ] Operating system an version: [enter here] (e.g Linux, macOS, Windows)
- [ ] **stlink tools version** and/or git commit hash: [enter here] (e.g v1.6.1/git-d0416149)
- [ ] stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [X] Target chip (and board, if applicable): [STM32F446] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
st-flash --area=option read
st-flash 1.7.0-184-g468b1d2
Failed to parse flash type or unrecognized flash type

detected chip_id parametres

# Device Type: STM32F446
# Reference Manual: RM0390
#
chip_id 0x421
flash_type 3
flash_size_reg 0x1fff7a22
flash_pagesize 0x20000
sram_size 0x20000
bootrom_base 0x1fff0000
bootrom_size 0x7800
option_base 0x40023c14
option_size 0x4
flags 2

2022-02-23T11:50:47 INFO common.c: STM32F446: 128 KiB SRAM, 512 KiB flash in at least 128 KiB pages.
0ffeaaed
```

Expected/description:

Introduced in this commit https://github.com/stlink-org/stlink/commit/e62b9e1f2a1b6dea7e9ccb053c14c2c6ad0b57d7 the option byte area has been changed. This leads to a read-out of 0x40023c14 which is somwhere in the AHB1 area.

[image]
[image]

Please change back option area to 0x1FFFC000 with a size of 16bytes, as shown in the datasheet and the attached images.

UserX404, 2022-02-23:
maybe dupe of https://github.com/stlink-org/stlink/issues/1156

Nightwalker-87, 2022-02-27:
It should be ok to leave this open, even if it's a duplicate.
We can mark this ticket respectively.

UserX404, 2022-02-27:
My thoughts about this issue were not correct.
I figured out 0x40023c14 (FLASH_OPTCT) is somehing like a shadow register of 0x1fffC000 and 0x1fffC800.
I just was focused to the option byte registers.
[image]

Nightwalker-87, 2022-12-29:
The fixed memory address in commit e62b9e1f2a1b6dea7e9ccb053c14c2c6ad0b57d7 appears to be correct.
Further details can be found e.g. on the ST community forum: [Link](https://community.st.com/s/question/0D50X0000AU4RuXSQV/on-what-address-do-the-option-bytes-on-stm32f446-reside-on).

closed 2022-12-29 by Nightwalker-87