stlink-org/stlink #1241: [STM32F4Discovery]: unknown chip id! 0x413

Catherinecyj, 2022-04-09:
Programmer/board type: Stlink /V2-onboard
Programmer firmware version: V2J37S26
Operating system: Linux (Ubuntu)
Stlink tools version and/or git commit hash: 1.7.0-186-gc4762e6
Stlink commandline tool name: st-flash, st-info
Target chip (and optional board): STM32F407VGT6U (STM32F4Discovery)

`st-flash write testvectors.bin 0x8000000` output:
```
st-flash 1.7.0-186-gc4762e6
/usr/local/stlink/chip: No such file or directory
2022-04-09 WARN common.c: unknown chip id! 0x413
Failed to connect to target
```

`st-info --probe` output:
```
/usr/local/stlink/chip: No such file or directory
Found 1 stlink programmers
    version: V2J37S26
    serial:    066DFF485157808667044350
    flash:    0
    sram:    0
    chipid:  0x413
```

However, I tried to use STM32 Cube Programmer and can recognized the device and the Device ID shown:
[image]
Also can program the .bin file successfully.

It seems like the device ID is 0x413 exactly, but how can solve this problem.

Catherine

Nightwalker-87, 2022-04-11:
Please use the issue template in the future - this ticket is actually invalid according to our project guidelines.

Duplicate of #1228.

bsalha1, 2022-11-22:
I am also getting this on my STM32F407VET6

Nightwalker-87, 2022-12-28:
This is not a chip-specific issue.
The output message `/usr/local/stlink/chip: No such file or directory` indicates a compilation bug regarding the installation path of the .chip configuration files used to detect MCU devices.

This bug has been fixed in commit f93adb92f2e4ecf05a9361cb723c98693586929d.
The correct installation path is `/usr/share/stlink/chips`.
In order to resolve this, please clean-up your installation with `make uninstall`, then `make clean`, preferably check out the latest commit on the `develop` branch and proceed with `make install`.

closed 2022-12-28 by Nightwalker-87