stlink-org/stlink #1244: `undefined symbol: libusb_set_option`

fabianpedd, 2022-04-14:
When trying to 
```
st-flash write generic_boot20_pc13.bin 0x08000000
```
I get 
```
st-flash 1.7.0
st-flash: symbol lookup error: /usr/lib/libstlink.so.1: undefined symbol: libusb_set_option
```
Same with 
```
st-info --probe
```
which also yields
```
st-info: symbol lookup error: /usr/lib/libstlink.so.1: undefined symbol: libusb_set_option
```

My setup:
`Ubuntu 18.04.6 LTS`  
`st-flash 1.7.0` installed from binary package (`stlink_1.7.0-1_amd64.deb`)

I ran `sudo apt-get install --reinstall libusb-1.0.0` with no success. 

There is a related [issue](https://github.com/stlink-org/stlink/issues/1089) but since it is closed I opened this new one.

jtmullen, 2022-05-31:
Did you ever find a solution to get this working?  I am seeing the same issue with both st-flash v1.6.1 and v1.7.0 on Mint 19.3 which also has Ubuntu 18.04 LTS as the upstream.

jtmullen, 2022-05-31:
Okay, actually I seem to have found the issue - it appears "libusb_set_option" was added in libusb 1.0.22 and Ubuntu 18.04 LTS ships with v1.0.21. It appears that libusb 1.0.22 came out before the first of this project, so it would seem that all the releases probably require this newer libusb.

fabianpedd, 2022-06-02:
Great, thanks. I will give it a try but seems logical to me.

Nightwalker-87, 2022-06-04:
Please leave ticket handling to Github Automation and Maintenance.