stlink-org/stlink #1253: [STM32L071]: Can't flash L071 device

rafaeldantaas, 2022-06-06:
- [x] Programmer/board type: [V2-clone]
- [x] Operating system an version: [macOS Monterey v12.2]
- [x] **stlink tools version** and/or git commit hash: [1.7.0]
- [x] stlink commandline tool name: [st-flash]
- [x] Target chip (and board, if applicable): [STM32L071]

I'm facing issues to flash a STM32L071 MCU. Found older closed tickets with same behavior, maybe this suppose to be fixed already in `v1.7.0`... Appreciate any help!

Command line output: 
```
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08000e00 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08000e80 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08000f00 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08000f80 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001000 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001080 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001100 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001180 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001200 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001280 erased
2022-06-06T16:58:26 INFO common.c: Flash page at addr: 0x08001300 erased
2022-06-06T16:58:26 INFO common.c: Finished erasing 39 pages of 128 (0x80) bytes
2022-06-06T16:58:26 INFO common.c: Starting Flash write for L0
2022-06-06T16:58:26 INFO common.c: Starting Half page flash write for STM32L core id
2022-06-06T16:58:26 INFO flash_loader.c: Successfully loaded flash loader in sram
2022-06-06T16:58:26 INFO flash_loader.c: Clear DFSR
2022-06-06T16:58:26 INFO common.c: Go to Thumb mode
2022-06-06T16:58:26 ERROR flash_loader.c: Write error
2022-06-06T16:58:26 WARN flash_loader.c: Loader state: R2 0x40 R15 0xFFFFFFFE
2022-06-06T16:58:26 WARN flash_loader.c: MCU state: DHCSR 0x103000B DFSR 0x8 CFSR 0x0 HFSR 0x0
2022-06-06T16:58:26 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2022-06-06T16:58:26 WARN common.c:
write_half_pages failed == -1
```

fdelbos, 2022-10-13:
I have the same error with an L011K4 nucleo

Nightwalker-87, 2023-01-02:
For debugging reasons this should be repeated with:
`st-flash --debug --reset write $(your binary name).bin 0x8000000`

Nightwalker-87, 2023-04-30:
@chenguokai Hi there. I've reviewed this issue and found that the observed behaviour is related to the flash-loader implementation you contributed in 2020 (which I believe is a very valuable contribution to this project).
Would you be interested to look into this topic and some obviously related issues?
I'd also appreciate to get in touch with you to find out more regarding the technical background of your implementation.

Nightwalker-87, 2023-04-30:
Duplicate of #681.

closed 2023-09-01 by Nightwalker-87