stlink-org/stlink #1257: [STM32F303]: build fails with gcc-12.1

robertgj, 2022-06-28:
- [x ] I made serious effort to avoid creating duplicate or nearly similar issue

- [x ] Programmer/board type: [STLINK/V2-onboard]
- [x ] Operating system an version: [Linux Fedora 36]
- [x ] **stlink tools version** and/or git commit hash: [commit 951859c119c570f3bdf8b42da0fbbaffec0848ab]
- [x ] stlink commandline tool name: [st-trace]
- [x ] Target chip (and board, if applicable): [STM32F303 (STMF3Discovery)]

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Command line output:

```
$ git clone https://github.com/stlink-org/stlink.git
Cloning into 'stlink'...
remote: Enumerating objects: 12328, done.
remote: Counting objects: 100% (2251/2251), done.
remote: Compressing objects: 100% (707/707), done.
remote: Total 12328 (delta 1580), reused 2135 (delta 1521), pack-reused 10077
Receiving objects: 100% (12328/12328), 16.67 MiB | 1.47 MiB/s, done.
Resolving deltas: 100% (8032/8032), done.
$ cd stlink
$ git status
On branch develop
Your branch is up to date with 'origin/develop'.

nothing to commit, working tree clean
$ cd stlink
$ make 
-- The C compiler identification is GNU 12.1.1
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Found Git: /usr/bin/git (found version "2.36.1") 
-- Rewrite xxxx/stlink/.version with 1.7.0!
-- stlink version: 1.7.0-195-g951859c
-- Major 1 Minor 7 Patch 0
-- Performing Test C_SUPPORTS_WALL
-- Performing Test C_SUPPORTS_WALL - Success
-- Performing Test C_SUPPORTS_WEXTRA
-- Performing Test C_SUPPORTS_WEXTRA - Success
-- Performing Test C_SUPPORTS_WSHADOW
-- Performing Test C_SUPPORTS_WSHADOW - Success
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2
-- Performing Test C_SUPPORTS_D_FORTIFY_SOURCE_2 - Success
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING
-- Performing Test C_SUPPORTS_FSTRICT_ALIASING - Success
-- Performing Test C_SUPPORTS_WUNDEF
-- Performing Test C_SUPPORTS_WUNDEF - Success
-- Performing Test C_SUPPORTS_WFORMAT
-- Performing Test C_SUPPORTS_WFORMAT - Success
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY
-- Performing Test C_SUPPORTS_WFORMAT_SECURITY - Success
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED
-- Performing Test C_SUPPORTS_WMAYBE_UNINITIALIZED - Success
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS
-- Performing Test C_SUPPORTS_WMISSING_VARIABLE_DECLARATIONS - Failed
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32
-- Performing Test C_SUPPORTS_WSHORTEN_64_TO_32 - Failed
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION
-- Performing Test C_SUPPORTS_WIMPLICIT_FUNCTION_DECLARATION - Success
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS
-- Performing Test C_SUPPORTS_WREDUNDANT_DECLS - Success
-- Performing Test C_SUPPORTS_FPIC
-- Performing Test C_SUPPORTS_FPIC - Success
-- Performing Test C_SUPPORTS_WERROR
-- Performing Test C_SUPPORTS_WERROR - Success
-- Found libusb: /usr/lib64/libusb-1.0.so  
[... 170 lines cut ...]
[ 69%] Building C object CMakeFiles/st-util.dir/src/st-util/gdb-remote.c.o
[ 71%] Building C object CMakeFiles/st-util.dir/src/st-util/gdb-server.c.o
[ 73%] Building C object CMakeFiles/st-util.dir/src/st-util/semihosting.c.o
[ 75%] Linking C executable bin/st-util
[ 75%] Built target st-util
[ 76%] Building C object CMakeFiles/st-trace.dir/src/st-trace/trace.c.o
[ 78%] Linking C executable bin/st-trace
[ 78%] Built target st-trace
[ 80%] Building C object src/stlink-gui/CMakeFiles/stlink-gui-local.dir/gui.c.o
[ 82%] Linking C executable ../../bin/stlink-gui-local
[ 82%] Built target stlink-gui-local
[ 84%] Building C object src/stlink-gui/CMakeFiles/stlink-gui.dir/gui.c.o
[ 86%] Linking C executable ../../bin/stlink-gui
[ 86%] Built target stlink-gui
[ 88%] Building C object tests/CMakeFiles/test-usb.dir/usb.c.o
[ 90%] Linking C executable ../bin/test-usb
[ 90%] Built target test-usb
[ 92%] Building C object tests/CMakeFiles/test-sg.dir/sg.c.o
[ 94%] Linking C executable ../bin/test-sg
[ 94%] Built target test-sg
[ 96%] Building C object tests/CMakeFiles/test-flash.dir/flash.c.o
[ 98%] Building C object tests/CMakeFiles/test-flash.dir/__/src/st-flash/flash_opts.c.o
[100%] Linking C executable ../bin/test-flash
[100%] Built target test-flash
```

robertgj, 2022-06-28:
This is better:
```
diff --git a/cmake/modules/c_flags.cmake b/cmake/modules/c_flags.cmake
index 44052bb..8b79838 100644
--- a/cmake/modules/c_flags.cmake
+++ b/cmake/modules/c_flags.cmake
@@ -21,6 +21,7 @@ add_cflag_if_supported("-Wall")
 add_cflag_if_supported("-Wextra")
 add_cflag_if_supported("-Wshadow")
 add_cflag_if_supported("-D_FORTIFY_SOURCE=2")
+add_cflag_if_supported("-O")
 add_cflag_if_supported("-fstrict-aliasing")
 add_cflag_if_supported("-Wundef")
 add_cflag_if_supported("-Wformat")
```

gszy, 2022-06-28:
>     -- Looking for sys/mman.h - not found

That was your original problem.

robertgj, 2022-06-28:
On Tue, 28 Jun 2022 03:36:57 -0700
Grzegorz Szymaszek ***@***.***> wrote:

> >     -- Looking for sys/mman.h - not found  
> 
> That was your original problem.
> 

Exactly. It was not found because the cmake check for sys/mman.h fails
because /usr/include/features.h warns if -D_FORTIFY_SOURCE=2 is defined
without optimisation.

From build/Release/CMakeFiles/CMakeError.log:

Determining if the include file sys/mman.h exists failed with the
following output: Change Dir:
xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/gmake -f Makefile cmTC_92d83/fast &&
gmake[1]: Entering directory
'xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp'
/usr/bin/gmake  -f CMakeFiles/cmTC_92d83.dir/build.make
CMakeFiles/cmTC_92d83.dir/build gmake[2]: Entering directory
'xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_92d83.dir/CheckIncludeFile.c.o
/usr/bin/cc   -Wall -Wextra -Wshadow -D_FORTIFY_SOURCE=2
-fstrict-aliasing -Wundef -Wformat -Wformat-security
-Wmaybe-uninitialized -Wimplicit-function-declaration -Wredundant-decls
-fPIC -Werror  -std=gnu11 -o
CMakeFiles/cmTC_92d83.dir/CheckIncludeFile.c.o -c
xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp/CheckIncludeFile.c
In file included from /usr/include/sys/mman.h:22, from
xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp/CheckIncludeFile.c:1:
/usr/include/features.h:412:4: error: #warning _FORTIFY_SOURCE requires
compiling with optimization (-O) [-Werror=cpp] 412 | #  warning
_FORTIFY_SOURCE requires compiling with optimization (-O) |    ^~~~~~~
cc1: all warnings being treated as errors gmake[2]: ***
[CMakeFiles/cmTC_92d83.dir/build.make:78:
CMakeFiles/cmTC_92d83.dir/CheckIncludeFile.c.o] Error 1 gmake[2]:
Leaving directory
'xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp'
gmake[1]: *** [Makefile:127: cmTC_92d83/fast] Error 2 gmake[1]: Leaving
directory
'xxxx/stlink-develop/build/Release/CMakeFiles/CMakeTmp'

ctsuzuki, 2022-07-28:
And

> I’m sorry, I wasn’t able to find the correlation between the `mman.h` error and the `_FORTIFY_SOURCES` issue, so I suspected missing development packages.
> 
> An explicit optimization flag had been set in the past, but the relevant code was removed in #1221.

And where I can find the missing packages, or who will put it back ?

gszy, 2022-07-29:
> And where I can find the missing packages, or who will put it back ?

I don’t have a Fedora at hand, but I guess the package in question is [glibc-devel](https://packages.fedoraproject.org/pkgs/glibc/glibc-devel/) or something like that. If it is already installed, then perhaps you are affected (#1262, for reference) by the issue described in the above comments. Please try reverting parts of #1221.

Nightwalker-87, 2022-08-27:
What is the current state here?

robertgj, 2022-08-27:
Hi,

See my comment:
https://github.com/stlink-org/stlink/issues/1257#issuecomment-1168114724

It appears that st-link is compiled with the default -O0 optimisation.
If you want -D_FORTIFY_SOURCE=2 then enable optimisation in
c_flags.cmake with -O. Otherwise remove -D_FORTIFY_SOURCE=2 from
c_flags.cmake.

Best regards,

Robert Jenssen

PR #1267 (2022-08-29): Fix compilation with gcc-12
Turn on compiler optimisation so that -D_FORTIFY_SOURCE=2 does not cause a warning from /usr/include/features.h
Closes #1257.
closed 2022-08-29 by Nightwalker-87