stlink-org/stlink #1263: [feature] Documentation inprovement

ApiumJacob, 2022-08-17:
- [X] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific request.

- [X] Programmer/board type: STLINK /V2
- [X] Operating system an version: Linux Ubuntu 20.04
- [X] **stlink tools version** and/or git commit hash: v1.7.0 (repo tag)
- [X] stlink commandline tool name: st-info + others
- [X] Target chip (and board, if applicable): STM32F446RE

While following the procedure build from source here: https://github.com/stlink-org/stlink/blob/develop/doc/compiling.md#linux I found the following error:

Commandline output:
```
jacob@linux:~/repos/stlink$ st-info --version
st-info: error while loading shared libraries: libstlink.so.1: cannot open shared object file: No such file or directory
```

Expected/description:
`v1.7.0`

Would it be inappropriate to add a step 4.5 to the Linux build instructions

4.5 `sudo ldconfig`

I was able to figure this out by searing for my error message in google which lead me to this page:

https://github.com/stlink-org/stlink/issues/478

This isn't this first time I've built from source and probably had this problem before and found the solution the same way.  But it would save 10-15 every time I have to build (once a year or so) if this was documented right in the install instructions.

Nightwalker-87, 2022-08-19:
I agree on that and will review the related instructions.
Actually this info shouldn't have vanished over time, but on the other hand I must also admit that I haven't run into this problem yet...

gszy, 2022-08-22:
Added in be66bbf200c718904514b044ba84d64a36456218, deleted in 8429ac078a783ee6d75db6bd8741eba732959c54. I don’t think ldconfig is necessary in every case, but I’m not sure when exactly it is.

ApiumJacob, 2022-08-22:
I know very little about ldconfig but my computer was a fresh install of the OS that never had stlink installed previously.  My theory is that since stlink had not been installed previously that the the necessary shared library was not in the cache.

commit 2926648b (2022-08-26): General Project Update

- Closes #1263.
-  Added Workflows for issue-management.
commit 43498ded (2022-08-27): General Project Update

- Closes #1263.
- Updated CHANGELOG.md
- Updated list of contributors
closed 2022-08-26 by Nightwalker-87