stlink-org/stlink #1272: [STM32F407]: st-trace can't reconnect

jared52005, 2022-09-28:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: [STLINK V2J29S7]
- [x] Operating system an version: [Windows 11 21H2 22000.978]
- [x] **stlink tools version** and/or git commit hash: [v1.7.0 Release date: 2021-04-25]
- [x] stlink commandline tool name: [`st-trace`]
- [x] Target chip (and board, if applicable): [STM32F407VET6]

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

**Step 1:**  
Begin reading a trace and then end the application using Ctrl+C
```
C:\STLink Tests>st-trace.exe
libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_8087&PID_0025\6&1143EA6&0&4'
libusb: info [winusb_get_device_list] libusb will not be able to access it
2022-09-28T17:42:11 INFO common.c: F4xx: 192 KiB SRAM, 1024 KiB flash in at least 16 KiB pages.
2022-09-28T17:42:11 INFO trace.c: Trace Port Interface configured to expect a 982 MHz system clock.
2022-09-28T17:42:11 INFO trace.c: Trace frequency set to 2000000 Hz.
2022-09-28T17:42:11 INFO trace.c: Reading Trace
Application is starting
^C
```
**Step 2:**  
 Reconnect. Sometimes `st-trace` is able to get through and it will reconnect correctly, usually I will end up with `LIBUSB_ERROR_TIMEOUT` error and then `LIBUSB_ERROR_PIPE` when I will pull USB cable out of ST-Link device
```
C:\STLink Tests>st-trace.exe
libusb: info [winusb_get_device_list] The following device has no driver: 'USB\VID_8087&PID_0025\6&1143EA6&0&4'
libusb: info [winusb_get_device_list] libusb will not be able to access it
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_VERSION
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_GET_CURRENT_MODE
[!] send_recv send request failed: LIBUSB_ERROR_PIPE
[!] send_recv STLINK_GET_CURRENT_MODE
libusb: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
[!] send_recv send request failed: LIBUSB_ERROR_IO
[!] send_recv STLINK_DEBUG_ENTER
libusb: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
[!] send_recv send request failed: LIBUSB_ERROR_IO
[!] send_recv STLINK_DEBUG_READCOREID
2022-09-28T17:42:24 ERROR common.c: Failed to read core_id
libusb: error [winusbx_submit_bulk_transfer] ReadPipe/WritePipe failed: [22] The device does not recognize the command.
[!] send_recv send request failed: LIBUSB_ERROR_IO
[!] send_recv STLINK_JTAG_READDEBUG_32BIT
2022-09-28T17:42:24 ERROR common.c: Can not connect to target. Please use 'connect under reset' and try again
2022-09-28T17:42:24 ERROR trace.c: Your stlink is not connected to a device
```
By sending termination signal using `ctrl+c` the `st-trace.exe` will be left hanging in the memory or it won't send ST-Link itself a signal to stop logging. This means that `st-trace` is unable to reconnect to the busy ST-Link. Pulling USB out of ST-Link and connecting it back will cause that it is possible to reconnect and log again.

**Expected action:** Some cleanup code, which would close / stop ST-Link logging before exiting the `st-trace.exe` program.

simplerobot, 2022-10-01:
Could you recreate the issue with verbose output using the following command?
 st-trace.exe -v

Could you give some information about libusb with the following command?
 apt-cache policy libusb-1.0*

gszy, 2022-10-01:
@simplerobot: if that’s Windows, there’s no apt-cache.

jared52005, 2022-10-02:
Here it is starting correctly in verbose mode. I also updated STLink via ST-Link Utility on firmware `V2J33S7` to see if it will resolve the issue. It did not, but now I need to specify clock on 168MHz during st-trace start
```
D:\stlink-org\bin>st-trace.exe -v --clock=168
2022-10-02T13:33:22 DEBUG trace.c: show_help = false
2022-10-02T13:33:22 DEBUG trace.c: show_version = false
2022-10-02T13:33:22 DEBUG trace.c: logging_level = 100
2022-10-02T13:33:22 DEBUG trace.c: core_frequency = 168000000 Hz
2022-10-02T13:33:22 DEBUG trace.c: trace_frequency = 0 Hz
2022-10-02T13:33:22 DEBUG trace.c: reset_board = true
2022-10-02T13:33:22 DEBUG trace.c: force = false
2022-10-02T13:33:22 DEBUG trace.c: serial_number = any
2022-10-02T13:33:23 DEBUG common.c: *** looking up stlink version
2022-10-02T13:33:23 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2022-10-02T13:33:23 DEBUG common.c: stlink pid     = 0x3748
2022-10-02T13:33:23 DEBUG common.c: stlink version = 0x2
2022-10-02T13:33:23 DEBUG common.c: jtag version   = 0x21
2022-10-02T13:33:23 DEBUG common.c: swim version   = 0x7
2022-10-02T13:33:23 DEBUG common.c: stlink current mode: dfu
2022-10-02T13:33:23 DEBUG usb.c: -- exit_dfu_mode
2022-10-02T13:33:23 DEBUG common.c: *** stlink_exit_dfu_mode ***
2022-10-02T13:33:23 DEBUG usb.c: JTAG/SWD freq set to 0
2022-10-02T13:33:23 DEBUG common.c: *** set_swdclk ***
2022-10-02T13:33:24 DEBUG common.c: stlink current mode: mass
2022-10-02T13:33:24 DEBUG common.c: *** stlink_enter_swd_mode ***
2022-10-02T13:33:24 DEBUG common.c: Loading device parameters....
2022-10-02T13:33:24 DEBUG common.c: *** stlink_core_id ***
2022-10-02T13:33:24 DEBUG common.c: core_id = 0x2ba01477
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x410fc241 at 0xe000ed00
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x10016413 at 0xe0042000
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x0400c000 at 0x1fff7a20
2022-10-02T13:33:24 INFO common.c: F4xx: 192 KiB SRAM, 1024 KiB flash in at least 16 KiB pages.
2022-10-02T13:33:24 DEBUG common.c: *** stlink_force_debug_mode ***
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x00001800 at 0xe0042008
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00001800 to 0xe0042008
2022-10-02T13:33:24 DEBUG common.c: *** stlink_reset ***
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x01030003 at 0xe000edf0
2022-10-02T13:33:24 DEBUG common.c: *** stlink_jtag_reset ***
2022-10-02T13:33:24 DEBUG common.c: *** stlink_jtag_reset ***
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x03010001 at 0xe000edf0
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0x01010001 at 0xe000edf0
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x01000000 to 0xe000edfc
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000002 to 0xe0002000
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001028
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001038
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001048
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001058
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0000000000 to 0xe0001000
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000027 to 0xe0042004
2022-10-02T13:33:24 DEBUG common.c: *** stlink_trace_enable ***
2022-10-02T13:33:24 DEBUG common.c: data_len = 2 0x2
 80 00
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000001 to 0xe0040004
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000053 to 0xe0040010
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000100 to 0xe0040304
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000002 to 0xe00400f0
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0xc5acce55 to 0xe0000fb0
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00000400 to 0xe0000e90
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x00010003 to 0xe0000e80
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0xffffffff to 0xe0000e00
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x0000000f to 0xe0000e40
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x400003ff to 0xe0001000
2022-10-02T13:33:24 DEBUG common.c: *** stlink_write_debug32 0x01000000 to 0xe000edfc
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_debug32 0xf52cb2ad at 0x00001fff
2022-10-02T13:33:24 INFO trace.c: Trace Port Interface configured to expect a 2464 MHz system clock.
2022-10-02T13:33:24 INFO trace.c: Trace frequency set to 2000000 Hz.
2022-10-02T13:33:24 INFO trace.c: Reading Trace
2022-10-02T13:33:24 DEBUG common.c: *** stlink_run ***
2022-10-02T13:33:24 DEBUG common.c: *** stlink_read_reg
2022-10-02T13:33:24 DEBUG common.c:  (16) ***
[... 70 lines cut ...]
2022-10-02T13:43:40 DEBUG common.c: *** stlink_enter_swd_mode ***
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_ENTER
2022-10-02T13:43:43 DEBUG common.c: Loading device parameters....
2022-10-02T13:43:43 DEBUG common.c: *** stlink_core_id ***
[!] send_recv send request failed: LIBUSB_ERROR_TIMEOUT
[!] send_recv STLINK_DEBUG_READCOREID
2022-10-02T13:43:46 ERROR common.c: Failed to read core_id
[... cut ...]
> apt-cache policy libusb-1.0*
I am on Windows, I unfortunately can't do that. however by searching through system32 I have found following
* libusb0.dll v1.2.6.0; `C:\Windows\System32\DriverStore\FileRepository\arduino_gemma.inf_amd64_ccc1c03de7b2c57b\amd64`
* libusb0.dll v1.2.6.0; `C:\Windows\System32\DriverStore\FileRepository\arduino_gemma.inf_amd64_ccc1c03de7b2c57b\x86`
* libusb0.dll v1.2.1.0; `C:\Windows\System32\DriverStore\FileRepository\pe_libusb_interface.inf_amd64_6ba29f16800301d5\redist`

Nightwalker-87, 2022-12-31:
@gszy @simplerobot Any ideas on how to proceed here?

simplerobot, 2022-12-31:
I think the problem is with the windows signal handling.  I think the fix is simple, but I don't have a good way to reproduce this issue or verify the fix.

Issue details:
I believe the [windows signal handler](https://github.com/stlink-org/stlink/blob/c7bcb5213f95a03b744aba8e76004fc1a4b6624f/src/st-trace/trace.c#L88) should return TRUE instead of FALSE to indicate that the signal was handled successfully.  By returning FALSE, the default handler gets called which calls ExitProcess before giving our process a chance to cleanup.

Nightwalker-87, 2022-12-31:
@simplerobot I'll have a look at it soon.
If you fancy, you may have a look at some other remaining issues. It would be great if you could help.
Some may be about to be solved, but need some additional input.

Nightwalker-87, 2023-01-01:
Ok, so we'd need someone to test this.
With a view of his GitHub-activity I suppose @jared52005 is no longer available...

jared52005, 2023-01-01:
I can test it, I will be available next weekend for it.

Nightwalker-87, 2023-01-02:
@jared52005 We believe this is fixed now.
Please check with the proposed PR submitted by @simplerobot which is linked to this issue.
We will wait for your testing before merging.

Nightwalker-87, 2023-01-08:
Ping @jared52005

jared52005, 2023-01-08:
Yes we can continue seems to work for me.

closed 2023-01-08 by Nightwalker-87