stlink-org/stlink #1273: [NUCLEO-WL55JC]: unknown chip id! 0x497

dmazzer, 2022-09-29:
This is a similar issue reported by https://github.com/stlink-org/stlink/issues/1241 but for a different target.

- Programmer/board type: Onboard STlink (NUCLEO-WL55JC)
- Operating system an version: Ubuntu 22.04
- **stlink tools version** and/or git commit hash: v1.7.0/git-179650d
- stlink commandline tool name: st-flash, st-info, STM32CubeProgrammer 
- Target chip (and board, if applicable): STM32WL55/NUCLEO-WL55JC

Commandline output:

```
$ st-flash --format ihex write build/FreeRTOS_Timers.hex.hex
st-flash 1.7.0
2022-09-29T14:32:12 INFO usb.c: Unable to match requested speed 1800 kHz, using 1600 kHz
2022-09-29T14:32:12 WARN common.c: unknown chip id! 0x497
Failed to connect to target

```
Commandline output:

```
$ st-info --probe
Found 1 stlink programmers
  version:    V3J10
  serial:     003000175553500920393256
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0497
```

With the STM32CubeProgrammer the board can be programmed without problems:

Commandline output:

```
$ ~/STM32Cube/STM32CubeProgrammer/bin/STM32_Programmer.sh -c port=SWD mode=UR -w build/FreeRTOS_Timers.hex -v -g
      -------------------------------------------------------------------
                        STM32CubeProgrammer v2.11.0                  
      -------------------------------------------------------------------

ST-LINK SN  : 003000175553500920393256
ST-LINK FW  : V3J10M3
Board       : NUCLEO-WL55JC
Voltage     : 3,26V
SWD freq    : 12000 KHz
Connect mode: Under Reset
Reset mode  : Hardware reset
Device ID   : 0x497
Revision ID : Rev Z
Device name : STM32WLxx
Flash size  : 256 KBytes
Device type : MCU
Device CPU  : Cortex-M4
BL Version  : 0xc3

Memory Programming ...
Opening and parsing file: FreeRTOS_Timers.hex
  File          : FreeRTOS_Timers.hex
  Size          : 13,34 KB 
  Address       : 0x08000000 

Erasing memory corresponding to segment 0:
Erasing internal memory sectors [0 6]
Download in Progress:
[==================================================] 100% 

File download complete
Time elapsed during download operation: 00:00:02.512

Verifying ...

Read progress:
[==================================================] 100% 

Download verified successfully 

RUNNING Program ... 
  Address:      : 0x8000000
Application is running, Please Hold on...
Start operation achieved successfully

```

Expected/description:
The expected result is that the st-flash could program the board.

gszy, 2022-09-30:
STM32WL is not supported in version 1.7.0, see <https://github.com/stlink-org/stlink/blob/develop/CHANGELOG.md>.

Nightwalker-87, 2022-10-23:
Edit: ... in the current `develop` branch. Thanks @gszy :+1: 
Thus this issue has already been resolved.

closed 2022-10-23 by Nightwalker-87