stlink-org/stlink #1274: [STLINK-V3MINIE]:  WARN usb.c: skipping ST device : 0x483:0x3754)

camojoe215, 2022-09-30:
- I am using a V3MINIE debugger on a STM32G474 target
- Using windows 10
- Using stlink-tools version 1.7.0
- When running st-info --probe

Commandline output:

```
st-info --probe
2022-09-30T16:19:57 WARN usb.c: skipping ST device : 0x483:0x3754)    
Found 0 stlink programmers
```
Should be expecting "Found 1 stlink programmer". Not sure why it's skipping considering that there is a definition accounting for PIDs using 3754 in the source code (usb.h). When trying this with a nucleo G474 it works as expected. One possible problem is that the V3MINIE uses usb-c?

camojoe215, 2022-09-30:
EDIT: I tried building the develop branch from source (this has the updated PID missing in the latest release). Building from source documentation isn't working either. CMake complains that libusb library cannot be found and there is no documentation addressing this.

stefandz, 2022-11-23:
I managed to get this working up to a point - built OK, modified [49-stlinkv3.rules](https://github.com/stlink-org/stlink/blob/develop/config/udev/rules.d/49-stlinkv3.rules) to include 

`# STLink V3MINIE
SUBSYSTEMS=="usb", ATTRS{idVendor}=="0483", ATTRS{idProduct}=="3754", \
    MODE:="0666", \
    SYMLINK+="stlinkv3_%n"
`

However, get the following when running `st-info --probe`

`Failed to parse flash type or unrecognized flash type
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V3J8
  serial:     000A0030544B500420343637
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
  dev-type:   unknown`

How can I help? Happy to provide a MINIE or even attempt a fix myself if there's a guide to where I should be looking to fix this :)

tanguy-s, 2022-12-06:
@stefandz have you managed to fix your issue ? 

I am running in something similar with an ST-Link V3 (idVendor=0483, idProduct=3753).

`st-info --probe`:

`Failed to parse flash type or unrecognized flash type
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V3
  serial:     0023003A3331511834333834
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
  dev-type:   unknown
`

Built from develop branch and supposedly have the correct udev for V3. 

Thank you

Nightwalker-87, 2022-12-22:
@angiglesias submitted PR #1281 for this issue.

closed 2022-12-28 by Nightwalker-87