stlink-org/stlink #1275: [STM32RET6]: Can‘t connect to target. After using mdk to flash.

cnjhb, 2022-10-28:
- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: [V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [Linux] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [1.7.0] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-flash] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [stm32ret6] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Commandline output:

```
[joseph@hikari stm32ret6usart]$ st-flash --connect-under-reset --format ihex write build/stm32ret6usart.hex
st-flash 1.7.0
2022-10-28T16:22:59 WARN common.c: NRST is not connected
2022-10-28T16:22:59 ERROR common.c: Soft reset failed: timeout
2022-10-28T16:22:59 ERROR common.c: Can not connect to target. Please use 'connect under reset' and try again
Failed to connect to target
```

Expected/description:

`Only after using mdk to flash`
` The stm32cube ide does not have this problem, and it can be fixed for our stlink. `

Ehsan2754, 2022-11-11:
Due to previous experience of mine here are my suggestions.
* double check connections
* double check boot pin configuration
* make sure you have already enabled SWD on your controller before flashing. try to erase current firmware on ur device somehow and program again with SWD enabled.

Nightwalker-87, 2023-01-02:
There is nothing to add to the feedback from @Ehsan2754 from my side.
This may likely be an hardware related issue.
Unfortunately we can't investigate this any further due to the limited amount of information available.
Please open a new ticket if the problem still persists; then with a full debug output (`st-flash --debug write`).

Nightwalker-87, 2023-01-02:
As appropriate advice has been given, we're closing this ticket now.

closed 2023-01-02 by Nightwalker-87