stlink-org/stlink #1276: [STLink-V23]: /usr/share//stlink/chips: No such file or directory

letshin, 2022-11-04:
Hi - I am not sure if these are two separate issues but after compiling from this repo I get the following error:
```
st-info --probe
/usr/share//stlink/chips: No such file or directory
Found 1 stlink programmers
  version:    V2J29S7
  serial:     50001000030000544334544E
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
```
Not sure if its related but while running PYOCD I get the following
```
pyocd load s140_nrf52_7.2.0_softdevice.hex --target nrf52833
0001139 C STLink error (9): Get IDCODE error [__main__]
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/pyocd/__main__.py", line 161, in run
    status = cmd.invoke()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/subcommands/load_cmd.py", line 96, in invoke
    with session:
  File "/usr/local/lib/python3.8/dist-packages/pyocd/core/session.py", line 391, in __enter__
    self.open()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/core/session.py", line 529, in open
    self._board.init()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/board/board.py", line 139, in init
    self.target.init()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/core/soc_target.py", line 147, in init
    seq.invoke()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/utility/sequencer.py", line 213, in invoke
    resultSequence.invoke()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/utility/sequencer.py", line 208, in invoke
    resultSequence = call()
  File "/usr/local/lib/python3.8/dist-packages/pyocd/coresight/dap.py", line 439, in _connect
    probe_conn.connect(self._protocol)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/coresight/dap.py", line 172, in connect
    self._connect_probe(protocol)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/coresight/dap.py", line 189, in _connect_probe
    self._probe.connect(protocol)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/probe/stlink_probe.py", line 159, in connect
    self._link.enter_debug(STLink.Protocol.SWD)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/probe/stlink/stlink.py", line 327, in enter_debug
    self._check_status(response)
  File "/usr/local/lib/python3.8/dist-packages/pyocd/probe/stlink/stlink.py", line 365, in _check_status
    raise exceptions.ProbeError(error_message)
pyocd.core.exceptions.ProbeError: STLink error (9): Get IDCODE error
```
---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [ ] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [ ] Operating system an version: [enter here] (e.g Linux, Windows)
- [ ] **stlink tools version** and/or git commit hash: [enter here] (e.g v1.6.1/git-d0416149)
- [ ] stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ ] Target chip (and board, if applicable): [enter here] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:

`short description of the expected value`

weiminshen99, 2022-11-13:
I am facing the same problem on MacOS (see the error message below):

% st-flash --reset write build/main.bin 0x8000000
st-flash 1.7.0-217-gf93adb9
/usr/share//stlink/chips: No such file or directory
Failed to enter SWD mode
Failed to connect to target
make: *** [flash] Error 255

How do I fix this problem?

weiminshen99, 2022-11-14:
Good news. I fixed the problem by changing one line in CMakeLists.txt, because new Mac does not like people to change /usr/share, but /usr/local/share is OK:

-# Define install directory /usr/share
-set(CMAKE_INSTALL_SHAREDIR /usr/share/)
+# Define install directory /usr/local/share [not /usr/share on MacOS]
+set(CMAKE_INSTALL_SHAREDIR /usr/local/share/)

The whole thing is now installed and working on my MacOS. I also write a paragraph in README for "Installation on Mac". Please let me know if you want that too.

letshin, 2022-11-14:
I was using it on linux, but didn't bother debugging. I instead returned it to Amazon and purchased a DAP_LINK adapter from eBay and use pyocd. This means I can't text your fix :(

closed 2022-12-18 by Nightwalker-87