stlink-org/stlink #1277: undefined symbol: libusb_set_option

PSLLSP, 2022-11-10:
Ubuntu 18.04.6

When I try to run st-info, I get an error:
`st-info: symbol lookup error: /usr/lib/libstlink.so.1: undefined symbol: libusb_set_option`

```
user@ubuntu:~$ sudo dpkg -i ~/Downloads/stlink_1.7.0-1_amd64.deb 
Selecting previously unselected package stlink.
(Reading database ... 386526 files and directories currently installed.)
Preparing to unpack .../stlink_1.7.0-1_amd64.deb ...
Unpacking stlink (1.7.0-1) ...
Setting up stlink (1.7.0-1) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2) ...
Processing triggers for bamfdaemon (0.5.3+18.04.20180207.2-0ubuntu1) ...
Rebuilding /usr/share/applications/bamf-2.index...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for man-db (2.8.3-2ubuntu0.1) ...

user@ubuntu:~$ st-info --version
v1.7.0

user@ubuntu:~$ st-info --probe
st-info: symbol lookup error: /usr/lib/libstlink.so.1: undefined symbol: libusb_set_option
```

gszy, 2022-11-13:
Is this the same as #1089 perhaps?

PSLLSP, 2022-11-13:
I think it is the same as #1089...

slyshykO, 2022-11-16:
looks like st-utils is built with a newer version of libusb that is present in the system.

try locally to build st-utils from the sources.

Nightwalker-87, 2022-12-28:
Fixed by #978.

closed 2022-12-28 by Nightwalker-87