stlink-org/stlink #1280: Build of stlink on macOS fails:

fxcoudert, 2022-11-20:
- [X] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: none, it's a compilation issue
- [x] Operating system an version: macOS 13 (Ventura)
- [x] **stlink tools version** and/or git commit hash: 1.7.0
- [x] stlink commandline tool name: no tool, it's a compilation issue
- [x] Target chip (and board, if applicable): none, it's a compilation issue

----

Compilation with `cmake` on macOS 13, using Xcode 14.1 (system compiler), leads to the following build failure:

```
Undefined symbols for architecture x86_64:
    "_SecTaskCopyValueForEntitlement", referenced from:
        _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
    "_SecTaskCreateFromSelf", referenced from:
        _darwin_detach_kernel_driver in libusb-1.0.a(darwin_usb.o)
```

These should be found in the Security framework, which should be linked with `-framework Security`.

amiiit, 2022-12-03:
On the same combination: Mac OS 13.0.1 I get the following error:

```
CMake Error at cmake_install.cmake:69 (file):
  file INSTALL cannot copy file
  "/Users/amit/workspace/webrelay/stlink/build/Release/lib/libstlink.a" to
  "/usr/local/lib/libstlink.a": Permission denied.
```

I did then try with `sudo make install` and I am getting now the following error:

```
CMake Error at cmake_install.cmake:137 (file):
  file cannot create directory: /usr/share/stlink/chips.  Maybe need
  administrative privileges.
```

Any idea if this is okay for the installer to create a directory over there?

Nightwalker-87, 2022-12-22:
@fxcoudert Please see https://github.com/stlink-org/stlink/issues/1269#issuecomment-1288165069.
Due to several restrictions introduced by Apple since 10.15 support for macOS will be dropped completely with the next release. Please refer to the official STMCube-Programmer by ST-Microelectronics instead.

closed 2022-12-22 by Nightwalker-87