stlink-org/stlink #1289: [STM32 L031K6]: st-flash hangs when trying to flash the binary file to the L031K6

fpi1337, 2022-12-22:
st-flash version: 1.7.0
OS: Ubuntu 22.04 LTS and Windows 10 Professional

Hello,

I just wanted to report an issue that happens for me when trying to flash something to my L031K6. Whenever I'll start the flashing process, I encounter a few warnings until a point where st-flash seems to be in an infinite loop. I tested it on Windows 10 Pro (msys2 st-flash and the minGW-32 release of this repo) as well as under Ubuntu 22.04 LTS. 

One note to take: if I try to write to the chip while using st-flash 1.6.1 (under my Debian/Mx Linux distribution) the writing process actually succeeds. However the output of the command line is still the same, except that st-flash actually "writes pages" after the `write_half_pages failed == -1`. 

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

- [x] Programmer/board type: [ST-LINK/V2-1 onboard] 
- [x] Operating system an version: [Ubuntu 22.04 LTS and W10 Pro] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [1.7.0] 
- [x] stlink commandline tool name: [st-flash]
- [x] Target chip (and board, if applicable): [STM32L031 (NUCLEO-L031K6)] 

Commandline output:

```
st-flash write main.bin 0x8000000
st-flash 1.7.0
2022-12-22T14:40:31 INFO common.c: L0xx Category 2: 8 KiB SRAM, 32 KiB flash in at least 128 byte pages.
file main.bin md5 checksum: 922bf8a7a875d08cfa64bf81e32e, stlink checksum: 0x0002598b
2022-12-22T14:40:31 INFO common.c: Attempting to write 3052 (0xbec) bytes to stm32 address: 134217728 (0x8000000)
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000000 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000080 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000100 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000180 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000200 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000280 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000300 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000380 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000400 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000480 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000500 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000580 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000600 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000680 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000700 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000780 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000800 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000880 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000900 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000980 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000a00 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000a80 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000b00 erased
2022-12-22T14:40:31 INFO common.c: Flash page at addr: 0x08000b80 erased
2022-12-22T14:40:31 INFO common.c: Finished erasing 24 pages of 128 (0x80) bytes
2022-12-22T14:40:31 INFO common.c: Starting Flash write for L0
2022-12-22T14:40:31 INFO common.c: Starting Half page flash write for STM32L core id
2022-12-22T14:40:31 INFO flash_loader.c: Successfully loaded flash loader in sram
2022-12-22T14:40:31 INFO flash_loader.c: Clear DFSR
2022-12-22T14:40:31 INFO common.c: Go to Thumb mode
2022-12-22T14:40:32 ERROR flash_loader.c: Flash loader run error
2022-12-22T14:40:32 WARN flash_loader.c: Loader state: R2 0x0 R15 0x0
2022-12-22T14:40:32 WARN flash_loader.c: MCU state: DHCSR 0x1080009 DFSR 0x0 CFSR 0x0 HFSR 0x0
2022-12-22T14:40:32 WARN common.c: l1_stlink_flash_loader_run(0x8000000) failed! == -1
2022-12-22T14:40:32 WARN common.c: 
write_half_pages failed == -1
```

Expected/description:
Success of the flashing process

It would be pretty nice if someone could help in that regard.

best regards,

Nightwalker-87, 2022-12-22:
Related to #681.

Nightwalker-87, 2022-12-29:
@BRA1L0R When looking at this however, one can assume that at least **v1.7.0** is still affected by #681.

closed 2023-09-01 by Nightwalker-87