stlink-org/stlink #1294: STM32H7A3: $Verification of Flash Failed

defyingphysics, 2023-01-06:
Hello, 

I have been trying to use the st-flash to upload a binary, but I am getting `ERROR common.c: Verification of flash failed at offset: 6144`. 
If I use Eclipse IDE to build and program, take the built .elf, convert to to bin using `arm-none-eabi-objcopy ` and then try to flash it works. But if I make a new build and try to flash that "raw" it fails every time. 

- [x] I made serious effort to avoid creating duplicate or nearly similar issue
- [x] Programmer/board type: [STLINK-V3E ] 
- [x] Operating system an version: [Linux] 
- [x] **stlink tools version** and/or git commit hash: [v1.7.0]
- [x] stlink commandline tool name: [st-flash write]
- [x] Target chip (and board, if applicable): [NUCLEO-H7A3ZI-Q] 

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:
Here is the debug output from a project after it has been built using Cube IDE. 

`arm-none-eabi-objcopy -O binary blinky.elf blinky.bin`
`st-flash --debug write blinky.bin 0x8000000`
```
st-flash 1.7.0
2023-01-06T22:15:27 INFO usb.c: Unable to match requested speed 1800 kHz, using 1000 kHz
2023-01-06T22:15:27 INFO common.c: H7Ax/H7Bx: 128 KiB SRAM, 2048 KiB flash in at least 8 KiB pages.
file blinky.bin md5 checksum: 8643ac75929d77f7548faabfc45914ab, stlink checksum: 0x002011c3
2023-01-06T22:15:27 INFO common.c: Attempting to write 21356 (0x536c) bytes to stm32 address: 134217728 (0x8000000)
2023-01-06T22:15:27 INFO common.c: Flash page at addr: 0x08000000 erased
2023-01-06T22:15:27 INFO common.c: Flash page at addr: 0x08002000 erased
2023-01-06T22:15:27 INFO common.c: Flash page at addr: 0x08004000 erased
2023-01-06T22:15:27 INFO common.c: Finished erasing 3 pages of 8192 (0x2000) bytes
2023-01-06T22:15:27 INFO common.c: Starting Flash write for H7
21356/21356 bytes written
2023-01-06T22:15:27 INFO common.c: Starting verification of write complete
2023-01-06T22:15:27 ERROR common.c: Verification of flash failed at offset: 6144
stlink_fwrite_flash() == -1

```
Debug Logs 
[logs_fail.txt](https://github.com/stlink-org/stlink/files/10363209/logs_fail.txt)

and then I use CubeIDE to program the aforementioned project on the STM32 and trying the exact same command works now.  
[logs_win.txt](https://github.com/stlink-org/stlink/files/10363210/logs_win.txt)

My ST Link version
[image]

I have tried the develop branch but it seems to have some bug currently:

```
❯ ./st-flash write ../../AudioBocs/blinky/build/blinky.bin 0x8000000
st-flash 1.7.0-233-gc7bcb52-dirty
/usr/share//stlink/chips: No such file or directory
2023-01-06T22:27:07 WARN common.c: unknown chip id! 0x480
Failed to connect to target
```

defyingphysics, 2023-01-06:
Sorry, this seems to definitely be a duplicate of #1249. Disabling UART pins allows me to use st-flash as normal. I let you decide if you this should be closed or if you would like some further help with debugging.

Nightwalker-87, 2023-01-06:
@defyingphysics Thanks for the note. Yes, please continue in the original ticket. This issue will remain as a duplicate.
We definitely need some help with debugging, because I don't have the respective hardware to do any testing.

Nightwalker-87, 2023-06-09:
I believe this issue is related to #1310.

Nightwalker-87, 2023-12-24:
Closed by #1249.

closed 2023-12-24 by Nightwalker-87