stlink-org/stlink #1297: st-util: symbol lookup error: st-util: undefined symbol: init_chipids

ayushman22git, 2023-01-24:
**Thank you for giving feedback to the stlink project.**
---
**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

- [ ] I made serious effort to avoid creating duplicate or nearly similar issue
 Have tried the solution mentioned on this link. [https://github.com/stlink-org/stlink/issues/1189] but the solution mentioned doesn't help in my case. Hence wanted to raise an issue - Please forgive if this is not acceptable. I am new to github and linux tools. 
In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- 

- [ ] [ ] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)

- [ ] Operating system an version: [Ubuntu 22.04 LTS] (e.g Linux, Windows)
- [ ] **stlink tools version** and/or git commit hash: [enter here] (e.g v1.6.1/git-d0416149)
- [ ] stlink commandline tool name: [st-info shows error - "st-info: symbol lookup error: st-info: undefined symbol: init_chipids"] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ ] Target chip (and board, if applicable): [STM32 NUCLEO-F746ZG] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Issue 1
Commandline output:

```
OUTPUT/ERROR of the commandline tool(s): No such file or directory
```

Expected/description:

libstlink.so files are present in the location /usr/lib32 . Can refer attached screenshots for more details which were asked in the earlier answer on the topic. 
[image]
[image]

Nightwalker-87, 2023-01-27:
You need to use the current HEAD of the the `develop` branch to include the latest tfixes.
Your version string shows that you used the package from the debian/ubuntu package sources.
We do not maintain these and they are out of date.

Nightwalker-87, 2023-03-19:
This issue can be traced back to an obviously faulty local installation of the toolset based on package sources.
The Debian Bookworm (Testing) package sources are also the base for the deb-package repo of Ubuntu 22.04 LTS. Thus the deb-packages of the stlink-toolset are identical. Having re-checked the repo-based installation of the related `stlink` packages on Debian (Testing) revealed that there is no compilation-related problem or bug here in the toolset itself.

In order to resolve this problem on the local instance, it is recommended to remove all leftovers from previous installations based on the list of installed files from the package info in the repository.
Afterwards a clean install either by compiling following the official project tutorial or a package install from the repository should succeed without any issues.

Against this background we can safely mark this issue as resolved.

closed 2023-03-19 by Nightwalker-87