stlink-org/stlink #1319: [feature] RESET_SOFT_AND_HALT as a specifiable behavior for monitor reset

sfarnswo, 2023-07-12:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific request.

- [x] Programmer/board type: `STLINK/V3`
- [x] Operating system an version: `Linux`
- [x] **stlink tools version** and/or git commit hash: f7fb0a6caaffdb31611b7f31bc4b5f04a04b7eeb
- [x] stlink commandline tool name: `st-util`
- [x] Target chip (and board, if applicable): `N/A`

I am proposing adding this option to be able to specify the a "halt" behavior for the "reset" monitor command.   
  
Using the Cortex-Debug extension on VsCode, and when debugging a target without the NRST pin connected, the default monitor reset attempts and succeeds at performing a soft reset in the absence of NRST. However; early breakpoints (such as those in `main` or `Reset_Handler`) do not trigger. This behavior is corrected when re-compiling the stlink-tools suite with the following modification:

``` diff

diff --git a/src/st-util/gdb-server.c b/src/st-util/gdb-server.c
index 9f9d938..5ad7087 100644
--- a/src/st-util/gdb-server.c
+++ b/src/st-util/gdb-server.c
@@ -1317,7 +1317,7 @@ int32_t serve(stlink_t *sl, st_state_t *st) {
                         reply = strdup("E00");
                     }
 
-                    ret = stlink_reset(sl, RESET_AUTO);
+                    ret = stlink_reset(sl, RESET_SOFT_AND_HALT);
                     if (ret) {
                         DLOG("Rcmd: reset failed with reset\n");
                         reply = strdup("E00");

```
Using the modified solution, the software reset succeeds and early breakpoints are triggered.

Nightwalker-87, 2023-09-01:
Obviously related to #1198.

Nightwalker-87, 2023-09-02:
Fixed in commit ae7c9a7dd789603d6c0c8be8696594dd1d2025e0.

commit a0c02161 (2023-09-14): Updated behaviour on Reset

Fixes an issue where early breakpoints did not trigger.
(Closes #1198) (Closes #1246) (Closes #1319)
PR #1332 (2023-09-16): Merge branch `testing` into `develop`
Formatting & style fixes.

[bugfix] Fixed flashing on STM32G0/G4 dual bank devices (Closes #1310)

[refactoring] Clean-up for stlink-lib

- Moved declarations for read/write functions to read_write.h .
- Checked & revised header includes
- Changed some datatypes for write_buffer_to_sram() to avoid explicit casting.

[bugfix] Fixed chip recovery after reset (Closes #1260)

[doc] Corrected compiling manual (Closes #1317)

[bugfix] Fixed an issue where early breakpoints did not trigger. (Closes #1198) (Closes #1246) (Closes #1319)

General Project Update

- U [... cut]
closed 2023-09-16 by Nightwalker-87