stlink-org/stlink #1320: STM32U575: Verification of flash failed at offset: 0

EdwardG323, 2023-07-12:
Hello,

I can't seem to get st-flash to work on my board. I keep getting the error stating the flash verification failed at offset 0. Se below for more details:

- [X] I made serious effort to avoid creating duplicate or nearly similar issue

- [X] Programmer/board type: STLINKV3 
- [X] Operating system an version: Ubuntu 23.04 
- [X] **stlink tools version** and/or git commit hash: 1.7.0-296-gf7fb0a6 
- [X] stlink commandline tool name: st-flash 
- [X] Target chip (and board, if applicable): NUCLEO-U575ZI-Q 

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

This is the error that i get after using the command: 
```
st-flash --reset write firmware.bin 0x8000000 
```

Commandline output:

```
2023-07-12T14:39:11 INFO common_flash.c: Starting verification of write complete
2023-07-12T14:39:11 ERROR common_flash.c: Verification of flash failed at offset: 0
2023-07-12T14:39:11 INFO common.c: Go to Thumb mode
stlink_fwrite_flash() == -1

```

Expected/description:

I expect my board to be programmed with my source code causing one of the LEDs to blink. I did not see a previous issue with my specific chip.

Nightwalker-87, 2023-07-15:
I have an idea for this, we have seen a similar problem with another board type recently and also have this board available for testing. Please give me some time for reviewing.

karstengit, 2023-07-30:
Maybe it is the same problem of verifying like in bug https://github.com/stlink-org/stlink/issues/1322 ?

closed 2023-11-23 by Nightwalker-87