stlink-org/stlink #1321: STM32G0B1: Erase fails starting page 64

vgeneves, 2023-07-21:
STM32G0B1 can have up to 512kB of flash memory with a page size of 2kB, which means up to 256 pages.

Looking at stlink_erase_flash_page in common_flash.c: 

```
    if (sl->flash_type == STM32_FLASH_TYPE_G0) {
      uint32_t flash_page = ((flashaddr - STM32_FLASH_BASE) / sl->flash_pgsz);
      stlink_read_debug32(sl, FLASH_Gx_CR, &val);
      // sec 3.7.5 - PNB[5:0] is offset by 3. PER is 0x2.
      val &= ~(0x3F << 3);
      val |= ((flash_page & 0x3F) << 3) | (1 << FLASH_CR_PER);
      stlink_write_debug32(sl, FLASH_Gx_CR, val);
    }
```
There is a problem with how is handled page number.

According to STM32G0B1 ref manual RM440 (https://www.st.com/resource/en/reference_manual/rm0444-stm32g0x1-advanced-armbased-32bit-mcus-stmicroelectronics.pdf) chapter 3.7.5 FLASH control register (FLASH_CR), page 105, PNB is 10 bit wide and not 6.

By masking the register value with 0x3F this limit to 64 the number of pages that can be erased.

Code should look like this :

```
    if (sl->flash_type == STM32_FLASH_TYPE_G0) {
      uint32_t flash_page = ((flashaddr - STM32_FLASH_BASE) / sl->flash_pgsz);
      stlink_read_debug32(sl, FLASH_Gx_CR, &val);
      // sec 3.7.5 - PNB[9:0] is offset by 3. PER is 0x2.
      val &= ~(0x3FF << 3);
      val |= ((flash_page & 0x3FF) << 3) | (1 << FLASH_CR_PER);
      stlink_write_debug32(sl, FLASH_Gx_CR, val);
    }
```

vgeneves, 2023-07-25:
Hi,

I also noticed that STM32G0 with a flash bigger than 128k have 2 flash banks.
When writing to bank2 BKER bit must be set and  page number is offseted by 256.

It is pretty well managed in zephyr os code: https://github.com/zephyrproject-rtos/zephyr/blob/main/drivers/flash/flash_stm32g0x.c#L119

Nightwalker-87, 2023-09-02:
@vgeneves The proposed fix will be submitted soon. I'm currently waiting for a PR-review before being able to proceed.
I'll let you know as soon as we are ready to test.

Nightwalker-87, 2023-09-16:
The PR has now merged. I've added the change locally, but are still reviewing some other fixes regarding dual bank devices which may also be part of it. The patch will be submitted to the `testing` branch shortly.

commit 78365edb (2023-09-16): Fixes for STM32H7 & STM32G0B1 devices

- Fixed flash lock for STM32H7 dual bank devices
- Fixed flash erase issue on STM32G0B1 (Closes #1321)
commit 7dcb1302 (2023-11-24): Fixes for STM32H7 & STM32G0B1 devices

- Fixed flash lock for STM32H7 dual bank devices
- Fixed flash erase issue on STM32G0B1 (Closes #1321)
closed 2023-11-24 by Nightwalker-87