stlink-org/stlink #1322: STM32F103C8T6 doesn't flash chinese clones

karstengit, 2023-07-30:
Please refer to https://github.com/stlink-org/stlink/issues/172 that is closed now

There are blue pills selled at Aliexpress that are assembled with STM32F103C8T6 cloned chips.
This chips have a different idcode and are not recognized by this stlink.

In the history (March 2020) I could find a modded version (st-flash 1.6.0-31-gf5d0454-dirty) that was able to flash this chips.
I don't have the source or a link any more for this version.

Now I buyed some new blue pills with an idential ID, but flashing fails in the verify section.

```
$ ./st-flash write EXAMPLE.bin 0x8000000
st-flash 1.6.0-31-gf5d0454-dirty
2023-07-30T12:43:04 INFO common.c: Loading device parameters....
2023-07-30T12:43:04 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2023-07-30T12:43:04 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
2023-07-30T12:43:04 INFO common.c: Ignoring 8 bytes of 0xff at end of file
2023-07-30T12:43:04 INFO common.c: Attempting to write 63800 (0xf938) bytes to stm32 address: 134217728 (0x8000000)
Flash page at addr: 0x0800f800 erased
2023-07-30T12:43:04 INFO common.c: Finished erasing 63 pages of 1024 (0x400) bytes
2023-07-30T12:43:04 INFO common.c: Starting Flash write for VL/F0/F3/F1_XL core id
2023-07-30T12:43:04 INFO flash_loader.c: Successfully loaded flash loader in sram
 63/63 pages written
2023-07-30T12:43:05 INFO common.c: Starting verification of write complete
2023-07-30T12:43:05 ERROR common.c: Verification of flash failed at offset: 0
stlink_fwrite_flash() == -1
```

I have no idea what is happening.
With the debug option it looks like:

```
2023-07-30T12:42:53 DEBUG common.c: *** stlink_read_debug32 1 is 0x40022010
2023-07-30T12:42:53 DEBUG common.c: *** stlink_write_debug32 81 to 0x40022010
 63/63 pages written
2023-07-30T12:42:53 INFO common.c: Starting verification of write complete
2023-07-30T12:42:53 DEBUG common.c: *** stlink_read_mem32 ***
data_len = 1024 0x400
 8c 03 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 37 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 0d 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 67 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 37 49 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 67 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 67 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff df 49 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff b7 4a ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 91 4b ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 5f 4c ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff a7 4d ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 67 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff 67 48 ff ff ff ff ff ff ff ff 00 00 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 52 41 ff ff ff ff ff ff ff ff 70 6d ff ff ff ff ff ff ff ff 90 48 ff ff ff ff ff ff ff ff 00 f0 ff ff ff ff ff ff ff ff 28 21 ff ff ff ff ff ff ff ff 88 48 ff ff ff ff ff ff ff ff 42 68 ff ff ff ff ff ff ff ff f6 e7 ff ff ff ff ff ff ff ff 16 00 ff ff ff ff ff ff ff ff 00 f0 ff ff ff ff ff ff ff ff 04 6d ff ff ff ff ff ff ff ff 61 26 ff ff ff ff ff ff ff ff 01 3b ff ff ff ff ff ff ff ff 00 f0 ff ff ff ff ff ff ff ff 04 6d ff ff ff ff ff ff ff ff 03 bd ff ff ff ff ff ff ff ff 01 d0 ff ff ff ff ff ff ff ff 01 62 ff ff ff ff ff ff ff ff 01 68 ff ff ff ff ff ff ff ff c1 61 ff ff ff ff ff ff ff ff 1e 21 ff ff ff ff ff ff ff ff 01 68 ff ff ff ff ff ff ff ff ff f7 ff ff ff ff ff ff ff ff 01 60 ff ff ff ff ff ff ff ff 01 69 ff ff ff ff ff ff ff ff 01 61 ff ff ff ff ff ff ff ff 41 6a ff ff ff ff ff ff ff ff 08 b5 ff ff ff ff ff ff ff ff 01 d2 ff ff ff ff ff ff ff ff 43 68 ff ff ff ff ff ff ff ff 4d 4e ff ff ff ff ff ff ff ff 08 bd ff ff ff ff ff ff ff ff 0c f8 ff ff ff ff ff ff ff ff 00 f0 ff ff ff ff ff ff ff ff ff f7 ff ff ff ff ff ff ff ff ff f7 ff ff ff ff ff ff ff ff ff f7 ff ff ff ff ff ff ff ff ff f7 ff ff ff ff ff ff ff ff 01 68 ff ff ff ff ff ff ff ff 47 f8 ff ff ff ff ff ff ff ff 51 ff ff ff ff ff ff ff ff ff 47 f8 ff ff ff ff ff ff ff ff 06 26 ff ff ff ff ff ff ff ff 07 d0 ff ff ff ff ff ff ff ff c9 00 ff ff ff ff ff ff ff ff 01 f8 ff ff ff ff ff ff ff ff 81 61 ff ff ff ff ff ff ff ff 01 60 ff ff ff ff ff ff ff ff 47 f8 ff ff ff ff ff ff ff ff 47 f8 ff ff ff ff ff ff ff ff 07 bd ff ff ff ff ff ff ff ff 01 6b ff ff ff ff ff ff ff ff 01 68 ff ff ff ff ff ff ff ff 15 d0 ff ff ff ff ff ff ff ff 00 29 ff ff ff ff ff ff ff ff 03 63 ff ff ff ff ff ff ff ff 04 6d ff ff ff ff ff ff ff ff c1 62 ff ff ff ff ff ff ff ff 1e 00 ff ff ff ff ff ff ff ff 1e 00 ff ff ff ff ff ff ff ff 00 20 ff ff ff ff ff ff ff ff 04 0d ff ff ff ff ff ff ff ff 00 20 ff ff ff ff ff ff ff ff 01 68 ff ff ff ff ff ff ff ff 06 20 ff ff ff ff ff ff ff ff 0b f8 ff ff ff ff ff ff ff ff 00 f0 ff ff ff ff ff ff ff ff c1 62 ff ff

2023-07-30T12:42:53 ERROR common.c: Verification of flash failed at offset: 0
2023-07-30T12:42:53 DEBUG common.c: *** stlink_read_debug32 ffff038c is 0x8000000
2023-07-30T12:42:53 DEBUG common.c: *** stlink_write_reg
data_len = 2 0x2
 80 00

2023-07-30T12:42:53 DEBUG common.c: *** stlink_read_debug32 ffffffff is 0x8000004
2023-07-30T12:42:53 DEBUG common.c: *** stlink_write_reg
data_len = 2 0x2
 80 00

2023-07-30T12:42:53 DEBUG common.c: *** stlink_run ***
stlink_fwrite_flash() == -1
2023-07-30T12:42:53 DEBUG common.c: *** stlink_exit_debug_mode ***
2023-07-30T12:42:53 DEBUG common.c: *** stlink_write_debug32 a05f0000 to 0xe000edf0
2023-07-30T12:42:53 DEBUG common.c: *** stlink_close ***
```

Trying a fresh downloaded unpatched Version 1.7 only shows

```
$ ./st-flash write EXAMPLE.bin 0x8000000
st-flash 1.7.0
2023-07-30T12:50:38 INFO common.c: Loading device parameters....
2023-07-30T12:50:38 INFO common.c: Device connected is: F1 Medium-density device, id 0x20036410
2023-07-30T12:50:38 INFO common.c: SRAM size: 0x5000 bytes (20 KiB), Flash: 0x20000 bytes (128 KiB) in pages of 1024 bytes
Failed to connect to target
```

Using openocd like described in the old bug above shows

```
$ openocd -f interface/stlink.cfg -f target/stm32f1x.cfg
Open On-Chip Debugger 0.12.0
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "hla_swd". To override use 'transport select <transport>'.
Info : The selected transport took over low-level target control. The results might differ compared to plain JTAG/SWD
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : STLINK V2J37S7 (API v2) VID:PID 0483:3748
Info : Target voltage: 3.225995
Warn : UNEXPECTED idcode: 0x2ba01477
Error: expected 1 of 1: 0x1ba01477
```

What can be done?

How the missing ID can be added to the source of version 1.7?

karstengit, 2023-07-30:
The modded version maybe helpful for others, so I attach it here.

[st-flash.zip](https://github.com/stlink-org/stlink/files/12208999/st-flash.zip)

```
$ file st-flash
st-flash: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, for GNU/Linux 2.6.32, BuildID[sha1]=ba79d72089a17b9d0e489828c6114efaccbefff8, not stripped

$ ldd st-flash
        linux-vdso.so.1 (0x00007fffbcbfe000)
        libstlink.so.1 => /lib/x86_64-linux-gnu/libstlink.so.1 (0x00007fac14000000)
        libusb-1.0.so.0 => /lib/x86_64-linux-gnu/libusb-1.0.so.0 (0x00007fac142eb000)
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007fac13e1f000)
        libudev.so.1 => /lib/x86_64-linux-gnu/libudev.so.1 (0x00007fac142bd000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007fac142b8000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fac14336000)
```

Nightwalker-87, 2023-07-31:
@karstengit: Please refer to our Tutorial on this issue.
There will be **_no support_** for these cloned chips.  They do break compatibility with other original STM32 chips in this toolset and we do not support nor want to deal with counterfeit products which are widely spread among these cloned devices.

closed 2023-07-31 by Nightwalker-87