stlink-org/stlink #1329: [NUCLEO-C031C6]: $device not recognized

vijaydhiman, 2023-08-23:
---
st-info --probe
Found 1 stlink programmers
  version:    V2J42S27
  serial:     066EFF555071494867244228
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x453
- [ ] Programmer/board type: [NUCLEO-C031C6] (STLINK V2 onboard)
- [ ] Operating system an version: [WSL] 
- [ ] **stlink tools version** and/or git commit hash: v1.7.0-296-gf7fb0a6
- [ ] stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-trace`)
- [ ] Target chip (and board, if applicable): [NUCLEO-C031C6] 

Commandline output:
st-info --probe
```
Found 1 stlink programmers
  version:    V2J42S27
  serial:     066EFF555071494867244228
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x453
```
in stmcubeide i'm able to flash and debug the device without any issue

---
tested the same command(st-info --probe) just to make sure the stlink works fine on my machine with nucleo f446re
Commandline output:
st-info --probe
```
Found 1 stlink programmers
  version:    V2J42S27
  serial:     066AFF545383564867215039
  flash:      524288 (pagesize: 131072)
  sram:       131072
  chipid:     0x421
  dev-type:   STM32F446
```

Nightwalker-87, 2023-08-25:
This device is not supported (yet).

vijaydhiman, 2023-08-26:
Is there any other way to flash it through cli?

Nightwalker-87, 2023-08-26:
No, not with this toolset currently. The correct handling of the hardware has to be implemented in the library first.
As you mentioned WSL on Windows, I recommend to use the official ST-Link Cube programmer from ST-Microelectronics instead.

closed 2023-11-23 by Nightwalker-87