stlink-org/stlink #1334: [feature] Provide access to the UART via virtual com port

srobertjames, 2023-09-04:
- [X ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific request.

- [ V2-1] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [ Linux Ubuntu] Operating system an version: [enter here] (e.g Linux, Windows)
- [ v1.7.0] **stlink tools version** and/or git commit hash: [enter here] (e.g v1.6.1/git-d0416149)
- [ st-util] stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ STM32F446RE NUCLEO-F446] Target chip (and board, if applicable): [enter here] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Is it possible to use st-link to access the virtual COM (serial) port provided by STLink? If so, how do I do that? I do not see any `/dev/ttyS` being created by st-util, and `tio` is not able to find it either.

Nightwalker-87, 2023-09-04:
No, this is currently not possible.

srobertjames, 2023-09-04:
I see.  What would be involved in implementing this? I may be able to help, but can find no docs on the ST-Link virtual com port implementation.

Nightwalker-87, 2023-09-04:
Well, one would need to deep dive into the toolset library codebase and find out how an implementation could look like while maintaining current functionality of this toolset. The other side of the coin is what you mentioned above, should it be publicly available. I'm afraid to say, that I can't provide any help on this topic.

Nightwalker-87, 2023-12-25:
@srobertjames I found [this link](https://electronics.stackexchange.com/questions/288187/is-it-possible-to-get-st-links-virtual-com-port-working-without-launching-any-p) that may help. As far as I can see, there is no need for any specific implementation.

neithernut, 2023-12-26:
My personal two cents:

...so they tell you to uninstall modemmanager, which is almost always a good idea. But depending on your distro you won't have that problem anyway.
Imo, the more relevant hint hidden in that stackexcahnge was the fact that you need the USB ACM driver.

Nightwalker-87, 2024-01-04:
@neithernut Have you been able to test the latter? Does this likely provide a solution from your point of view?

neithernut, 2024-01-04:
> @neithernut Have you been able to test the latter? Does this likely provide a solution from your point of view?

I've confirmed that, if I plug in my (dated) STM32F411 nucleo board, the device does show up (according to `dmesg`) with the USB ACM driver. I planned playing around with it a bit on the weekend. When I do, I'll report back here, maybe with proposed changes to documentation.

neithernut, 2024-01-05:
Update: got to play around with my board, and everything works just fine. The device pops up as `/dev/ttyACM0` on my machine and at least 9600 8N1 communication works both ways (I used `minicom` and a simple toy example on the nucleo board).

But now I struggle a bit with where to put this information. Serial communication doesn't concern any of the tools provided in this repo. Maybe we could add some hints to `doc/tutorial.md`? Maybe a new file in `doc/`? Will users even look there?

Nightwalker-87, 2024-01-06:
@neithernut Thanks for the update. I'll think about how to deal with the documentation-related part soon.

Nightwalker-87, 2024-01-20:
@neithernut We can add a new section to our `tutorial.md` under "Solutions to common problems" and extend the title to "How-To's & Solutions to common problems" or anything alike.

Nightwalker-87, 2024-01-21:
I've added a description to our project tutorial in order to resolve this issue which is now closed.

commit 32e8dcc8 (2024-01-21): [doc] Updated tutorial: Access to the UART via a virtual COM port

(Closes #1334)
closed 2024-01-21 by Nightwalker-87