stlink-org/stlink #1335: [STM32F446]: Target voltage too low for 32-bit flash, using 8-bit flash writes

yhuangece7, 2023-09-09:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: [ST-LINK V3] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x ] Operating system an version: [Linux Ubuntu 22.04] (e.g Linux, Windows)
- [x ] **stlink tools version** and/or git commit hash: [V1.7.0] (e.g v1.6.1/git-d0416149)
- [X ] stlink commandline tool name: [st-flash] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [X ] Target chip (and board, if applicable): [STM32F446] (e.g STM32F103C8T6 (NUCLEO-F103RB))

When I used st-flash to download an elf file to my STM32F446 device, I saw the error message "Target voltage (1 mV) too low for 32-bit flash, using 8-bit flash writes". Although I can see the final message "Flash written and verified! jolly good!" at the end of the flash command, I can verify the FW download is not really successful after doing some experiments. Since the elf file I used can be flashed to the same device using ST-Cube-programmer installed in the same Ubuntu computer, I suspect the not successful flashing is somewhat related to the reported error.

And tracing it back to the message, we can see it is from line 631 of /src/stlink-lib/flash_loader.c. And I also found a similar issue reported in https://github.com/stlink-org/stlink/issues/1264 , but that is a feature request and I did not see a solution or a real problem reported there. 

```
    if (sl->flash_type == STM32_FLASH_TYPE_L4) {
      // L4 does not have a byte-write mode
      if (voltage < 1710) {
        ELOG("Target voltage (%d mV) too low for flash writes!\n", voltage);
        return (-1);
      }
    } else {
      if (voltage > 2700) {
        ILOG("enabling 32-bit flash writes\n");
        write_flash_cr_psiz(sl, 2, BANK_1);
      } else {
        ILOG("Target voltage (%d mV) too low for 32-bit flash, using 8-bit flash writes\n", voltage);
        write_flash_cr_psiz(sl, 0, BANK_1);
      }
    }
```

Since this error is the only one I can see, and I do not know how this can happen, or a successful flashing using official ST-CUBE-PROGRAMMER installed in Ubuntu does have this issue or not, I want to see how to fix it or bypass it. If anyone can direct me to some actions I can try, please advise.

Thank you!

Nightwalker-87, 2023-09-14:
Duplicate of #1335.

Nightwalker-87, 2023-09-18:
I currently have no idea on how to technically deal with this issue.
Is there anyone else who can look at this?

closed 2023-12-03 by Nightwalker-87