stlink-org/stlink #1340: stm32f411re: Flash programming error 0x40

sparkdragon586, 2023-10-07:
**Thank you for giving feedback to the stlink project.**

---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: ST-link-V2 (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [arch up to date] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [V1.7] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-flash] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [blackpill] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
❯ st-flash --debug write tmp.bin 0x08000000
st-flash 1.7.0
2023-10-06T21:01:32 DEBUG common.c: *** looking up stlink version
2023-10-06T21:01:32 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2023-10-06T21:01:32 DEBUG common.c: stlink pid     = 0x3748
2023-10-06T21:01:32 DEBUG common.c: stlink version = 0x2
2023-10-06T21:01:32 DEBUG common.c: jtag version   = 0x2b
2023-10-06T21:01:32 DEBUG common.c: swim version   = 0x7
2023-10-06T21:01:32 DEBUG common.c: stlink current mode: dfu
2023-10-06T21:01:32 DEBUG usb.c: -- exit_dfu_mode
2023-10-06T21:01:32 DEBUG common.c: *** stlink_exit_dfu_mode ***
2023-10-06T21:01:32 DEBUG usb.c: JTAG/SWD freq set to 0
2023-10-06T21:01:32 DEBUG common.c: *** set_swdclk ***
2023-10-06T21:01:33 DEBUG common.c: stlink current mode: mass
2023-10-06T21:01:33 DEBUG common.c: *** stlink_enter_swd_mode ***
2023-10-06T21:01:33 DEBUG common.c: Loading device parameters....
2023-10-06T21:01:33 DEBUG common.c: *** stlink_core_id ***
2023-10-06T21:01:33 DEBUG common.c: core_id = 0x2ba01477
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0x410fc241 at 0xe000ed00
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0x10006431 at 0xe0042000
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0x0200c000 at 0x1fff7a20
2023-10-06T21:01:33 INFO common.c: stm32f411re: 128 KiB SRAM, 512 KiB flash in at least 16 KiB pages.
2023-10-06T21:01:33 DEBUG common.c: stlink current mode: debug (jtag or swd)
2023-10-06T21:01:33 DEBUG common.c: stlink current mode: debug (jtag or swd)
2023-10-06T21:01:33 DEBUG common.c: *** stlink_force_debug_mode ***
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0xe0042008
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0x00001800 to 0xe0042008
2023-10-06T21:01:33 DEBUG common.c: *** stlink_status ***
2023-10-06T21:01:33 DEBUG usb.c: core status: 03030003
2023-10-06T21:01:33 DEBUG common.c:   core status: halted
file tmp.bin md5 checksum: 6ae59e6485377ee5470c854761551ea, stlink checksum: 0x000ff000
2023-10-06T21:01:33 INFO common.c: Attempting to write 4096 (0x1000) bytes to stm32 address: 134217728 (0x8000000)
2023-10-06T21:01:33 DEBUG common.c: *** stlink_core_id ***
2023-10-06T21:01:33 DEBUG common.c: core_id = 0x2ba01477
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x40023c0c
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0x000000f0 to 0x40023c0c
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0x80000000 at 0x40023c10
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0x45670123 to 0x40023c04
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0xcdef89ab to 0x40023c04
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x40023c10
2023-10-06T21:01:33 DEBUG common.c: Successfully unlocked flash
EraseFlash - Sector:0x0 Size:0x4000 2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0000000000 at 0x40023c10
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0x00000002 to 0x40023c10
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0x00000002 at 0x40023c10
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0x00010002 to 0x40023c10
[... 75 lines cut ...]
2023-10-06T21:01:33 ERROR common.c: Flash programming error: 0x00000040
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0xffffffff at 0x08000000
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_reg
2023-10-06T21:01:33 DEBUG common.c: data_len = 2 0x2
 0d 00
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_debug32 0xffffffff at 0x08000004
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_reg
2023-10-06T21:01:33 DEBUG common.c: data_len = 2 0x2
 80 00
2023-10-06T21:01:33 DEBUG common.c: *** stlink_run ***
2023-10-06T21:01:33 DEBUG common.c: *** stlink_read_reg
2023-10-06T21:01:33 DEBUG common.c:  (16) ***
2023-10-06T21:01:33 DEBUG common.c: data_len = 8 0x8
 80 00 00 00 03 00 00 61
2023-10-06T21:01:33 DEBUG usb.c: r_idx (16) = 0x61000003
stlink_fwrite_flash() == -1
2023-10-06T21:01:33 DEBUG common.c: *** stlink_exit_debug_mode ***
2023-10-06T21:01:33 DEBUG common.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2023-10-06T21:01:33 DEBUG common.c: *** stlink_close ***

```

Expected/description:

st-flash and stlink-gui fail to write flash (they erase) no matter what I try. st-info shows valid info. Sorry if this is super simple or not enough info. feel free to ask for more info.

Nightwalker-87, 2023-12-25:
Can someone with a blackpill board (STM32F411RE) please try to reproduce this issue?

darkmoon32, 2024-01-03:
@Nightwalker-87 I do not have blackpill board but I have tested it with Nucleo-F411RE if that helps.  I was not able to reproduce the issue. Perhaps it is a HW issue because the boards' voltage is a little low.
[st-flash.txt](https://github.com/stlink-org/stlink/files/13816383/st-flash.txt)

Nightwalker-87, 2024-01-04:
@darkmoon32 Thank you very much for reporting back. :wink: 
The fact that the issue is not reproducible with the same chip is good news.
It indicates that there is no general issue with the toolset itself related to this reporting.
Thus we shall be fine to close this ticket as resolved and verified.
One may open a new issue anytime, if a hardware issue can be definitely ruled out.

closed 2024-01-04 by Nightwalker-87