stlink-org/stlink #1342: STM32H755ZI-Q: Flash write fails on second page (0x08100000)

starkedem, 2023-10-14:
- Programmer/board type: STLINK V3 / NUCLEO-H755ZI-Q
- Operating system an version:: Win11
- stlink tools version and/or git commit hash: 1.7.0-314-g40fbdb4
- stlink commandline tool name: st-flash
- Target chip (and board, if applicable): NUCLEO-H755ZI-Q

Commandline output:
```
st-flash --flash=2M --freq=1000k --connect-under-reset --reset --debug write ./Build/firmware.bin 0x08100000
st-flash 1.7.0-314-g40fbdb4
2023-10-13T18:26:28 DEBUG common.c: *** looking up stlink version ***
2023-10-13T18:26:28 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2023-10-13T18:26:28 DEBUG common.c: stlink pid     = 0x374e
2023-10-13T18:26:28 DEBUG common.c: stlink version = 0x3
2023-10-13T18:26:28 DEBUG common.c: jtag version   = 0xd
2023-10-13T18:26:28 DEBUG common.c: swim version   = 0x0
2023-10-13T18:26:28 DEBUG common.c: stlink current mode: mass
2023-10-13T18:26:28 DEBUG usb.c: JTAG/SWD freq set to 1000
2023-10-13T18:26:28 DEBUG common.c: *** stlink_enter_swd_mode ***
2023-10-13T18:26:28 DEBUG common.c: *** stlink_jtag_reset 0 ***
2023-10-13T18:26:28 DEBUG common.c: *** stlink_jtag_reset 1 ***
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x02030003 at 0xe000edf0
2023-10-13T18:26:28 DEBUG common.c: *** stlink_soft_reset (halt) ***
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0xa05f0003 to 0xe000edf0
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0x01000501 to 0xe000edfc
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0x00000008 to 0xe000ed30
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x00030003 at 0xe000edf0
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0x05fa0004 to 0xe000ed0c
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x02010001 at 0xe000edf0
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x02030003 at 0xe000edf0
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x00030003 at 0xe000edf0
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x00000009 at 0xe000ed30
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0x0000001f to 0xe000ed30
2023-10-13T18:26:28 DEBUG common.c: stlink current mode: debug (jtag or swd)
2023-10-13T18:26:28 DEBUG common.c: Loading device parameters....
2023-10-13T18:26:28 DEBUG common.c: *** stlink_core_id ***
2023-10-13T18:26:28 DEBUG common.c: core_id = 0x6ba02477
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x411fc271 at 0xe000ed00
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x20036450 at 0x5c001000
2023-10-13T18:26:28 DEBUG chipid.c: detected chip_id parameters

2023-10-13T18:26:28 DEBUG chipid.c: # Device Type: STM32H74x_H75x
2023-10-13T18:26:28 DEBUG chipid.c: # Reference Manual: RM0433
2023-10-13T18:26:28 DEBUG chipid.c: #
2023-10-13T18:26:28 DEBUG chipid.c: chip_id 0x450
2023-10-13T18:26:28 DEBUG chipid.c: flash_type 7
2023-10-13T18:26:28 DEBUG chipid.c: flash_size_reg 0x1ff1e880
2023-10-13T18:26:28 DEBUG chipid.c: flash_pagesize 0x20000
2023-10-13T18:26:28 DEBUG chipid.c: sram_size 0x20000
2023-10-13T18:26:28 DEBUG chipid.c: bootrom_base 0x1ff00000
2023-10-13T18:26:28 DEBUG chipid.c: bootrom_size 0x20000
2023-10-13T18:26:28 DEBUG chipid.c: option_base 0x5200201c
2023-10-13T18:26:28 DEBUG chipid.c: option_size 0x2c
2023-10-13T18:26:28 DEBUG chipid.c: flags 3

2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0x00000800 at 0x1ff1e880
2023-10-13T18:26:28 INFO common.c: STM32H74x_H75x: 128 KiB SRAM, 2048 KiB flash in at least 128 KiB pages.
2023-10-13T18:26:28 DEBUG common.c: *** stlink_force_debug_mode ***
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x5c001054
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0x00040000 to 0x5c001054
2023-10-13T18:26:28 DEBUG common.c: *** stlink_status ***
2023-10-13T18:26:28 DEBUG usb.c: core status: 00030003
2023-10-13T18:26:28 DEBUG common.c:   core status: halted
file ./Build/firmware.bin md5 checksum: eb1ad0f9fab3b6a2bf1371dc3225ce, stlink checksum: 0x00016987
2023-10-13T18:26:28 INFO common_flash.c: Attempting to write 1468 (0x5bc) bytes to stm32 address: 135266304 (0x8100000)
2023-10-13T18:26:28 DEBUG common.c: *** stlink_core_id ***
2023-10-13T18:26:28 DEBUG common.c: core_id = 0x6ba02477
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x52002010
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x52002110
2023-10-13T18:26:28 DEBUG read_write.c: *** stlink_write_debug32 0x000e0000 to 0x52002010
[... 324 lines cut ...]
1468/1468 bytes written
2023-10-13T18:26:30 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x52002010
2023-10-13T18:26:30 DEBUG read_write.c: *** stlink_read_debug32  0x00050000 at 0x52002110
2023-10-13T18:26:30 ERROR common_flash.c: Flash programming error: 0x00040000
2023-10-13T18:26:30 DEBUG read_write.c: *** stlink_read_debug32  0xffffffff at 0x08100004
2023-10-13T18:26:30 DEBUG read_write.c: *** stlink_write_reg
2023-10-13T18:26:30 DEBUG common.c: *** stlink_run ***
2023-10-13T18:26:30 DEBUG read_write.c: *** stlink_read_reg
2023-10-13T18:26:30 DEBUG read_write.c:  (16) ***
2023-10-13T18:26:30 DEBUG common.c: data_len = 8 0x8
 80 00 00 00 00 00 00 01
2023-10-13T18:26:30 DEBUG usb.c: r_idx (16) = 0x01000000
stlink_fwrite_flash() == -1
2023-10-13T18:26:30 DEBUG common.c: *** stlink_exit_debug_mode ***
2023-10-13T18:26:30 DEBUG read_write.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2023-10-13T18:26:30 DEBUG common.c: *** stlink_close ***
make: *** [../rules.mk:103: flash] Error -1
```

Expected/description:
Unable to load firmware at the 0x08100000 address.

Release version can neither erase or write flash with address greater than 0x08100000. It doesn't return any error, just flash verification fails. If you download the firmware using ST-LINK Utility prior to using st-flash, it doesn't even return verification error, because *.bin file and flash match (no thanks to st-flash).

"Last commit" version can erase sectors after 0x08100000 but can not write anything.

Nightwalker-87, 2023-10-15:
Duplicate of #1146.

Nightwalker-87, 2023-10-15:
This issue has been fixed already.

closed 2023-10-15 by Nightwalker-87