stlink-org/stlink #1357: [ST-Link V3_MiniE]: LIBUSB_ERROR_TIMEOUT only when debugger is powered without connection to target board

OSHW-Rico, 2023-11-16:
- V3MiniE
- Ubuntu 20.x
- st-flash 1.7.0-296-gf7fb0a6-dirty -> dirty because I have added debug lines for something else (will try with stock version again)
- `st-flash --connect-under-reset --reset --format ihex --area=main write firmware.hex`
- STM32G0Bx_G0Cx
Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
st-flash 1.7.0-296-gf7fb0a6-dirty
2023-11-16T11:46:00 ERROR usb.c: GET_VERSION send request failed: LIBUSB_ERROR_TIMEOUT
2023-11-16T11:46:03 ERROR usb.c: GET_CURRENT_MODE send request failed: LIBUSB_ERROR_TIMEOUT
2023-11-16T11:46:06 ERROR usb.c: DRIVE_NRST send request failed: LIBUSB_ERROR_TIMEOUT
^C
➜  ~ st-info --probe
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V3
  serial:     003400253132510838363431
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
  dev-type:   unknown
```

Expected/description:

If I do not connect the debugger to my boards and power it via USB, the issues arises (sometimes). It's not really reproducible. Sometimes it is the other way around. Since the ST-Link is behind a custom USB-Hub I don't want to blame st-flash! But I am wondering what the issue could be. When are `LIBUSB_ERROR_TIMEOUT` expected? I guess that could be up related to the USB-Hub?

I will try and see if I can reproduce the issue without the USB-Hub.

Nightwalker-87, 2023-12-23:
@OSHW-Rico  Any update on this?

Nightwalker-87, 2024-01-15:
@OSHW-Rico  Something appears to be not ok with the detection of your V3MiniE programmer.
I've the same model here and tried to reproduce the issue under most equal conditions.
As can be seen one should receive a complete (firmware) version string:

```
$ st-info --probe
Failed to enter SWD mode
Found 1 stlink programmers
  version:    V3J13        <-----
  serial:     003700253431************ (<--- intentionally blacked out)
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
  dev-type:   unknown
```

Please reinstall your libusb library, ensure it is the latest available version and also update the firmware of your STLINK-V3MINIE to the most recent revision.

Nightwalker-87, 2024-01-15:
No evidence for a toolset-related issue could be found after intensive testing and the issue could not be reproduced.
Nevertheless appropriate advice has been given for further investigation of the used hardware.
Thus I'm closing this issue now as resolved.

closed 2024-01-15 by Nightwalker-87