stlink-org/stlink #1359: STM32G0B1RE / nucleo_g0b1re: False chip id reading

SamyFrancelet, 2023-11-20:
- Programmer/board type: [Stlink V2-onboard]
- Operating system and version: [Windows 10 / Windows 11 / Arch Linux]
- Stlink tools version and/or git commit hash: [v1.7.0]
- Stlink commandline tool name: [st-util / st-info / st-flash]
- Target chip (and board if applicable): [STM32G0B1RE / nucleo_g0b1re]

Commandline output:

```
> st-util
st-util
2023-11-20T20:18:32 WARN common.c: unknown chip id! 0x90
2023-11-20T20:18:32 INFO gdb-server.c: Listening at *:4242...
```

```
> st-info --probe
Found 1 stlink programmers
  version:    V2J43S28
  serial:     066BFF485671664867182946
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0090
```

```
> st-flash erase
st-flash 1.7.0
2023-11-20T20:17:44 WARN common.c: unknown chip id! 0x90
Failed to connect to target
```

Target information from STM32CubeProgrammer (on Windows 10):
```
20:19:37 : ST-LINK SN  : 066BFF485671664867182946
20:19:37 : ST-LINK FW  : V2J43M28
20:19:37 : Board       : NUCLEO-G0B1RE
20:19:37 : Voltage     : 3.24V
20:19:37 : SWD freq    : 4000 KHz
20:19:37 : Connect mode: Normal
20:19:37 : Reset mode  : Software reset
20:19:37 : Device ID   : 0x467
20:19:37 : Revision ID : Rev A
```

Description:
The chip id is incorrect.

This problem only happens on this nucleo_g0b1 board, I have other boards (F411RE / LoRaWAN boards, etc..) and the chip id is read properly. I unfortunately doesn't have access to other G0 boards to probe if the problem happens on other Cortex-M0+ cores.
I've tested this only with the lastest release (v1.7.0), on multiple machines (Windows 10/11, Arch Linux) and got the same problem everywhere.

PR #1126 doesn't seem to fix my issue as it should be the v1.7.0 Release

I've ensured that the STLink is properly updated, and I can also program the board properly using STM32CubeProgrammer or STM32CubeIDE.

Is there anything else that I can do to help solve this issue ?

Thanks for your support,
/samy

73jn, 2023-11-23:
Hi, I have the same problem on a custom board with a STM32G0B0RE using an ST-Link V2, STLink tools version 1.7.0 on Windows 10 64-bits:
```
Found 1 stlink programmers
  version:    V2J43S7
  serial:     38FF6D06304E4B3029341243
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0467
```

```
st-util
2023-11-23T15:16:12 WARN common.c: unknown chip id! 0x467
2023-11-23T15:16:12 INFO gdb-server.c: Listening at *:4242...
```

I don't have this problem on any other board. The chip id is correct but STLink tools version 1.7.0 doesn't recognise it.

Thanks for your support !

Nightwalker-87, 2023-12-23:
Fixed in #1140.

closed 2023-12-23 by Nightwalker-87