stlink-org/stlink #1360: [STM32F103] stlink programmers are disappearing after running probing twice.

agicquel, 2023-11-24:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [enter here] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [enter here] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [enter here] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Hi, I want to connect multiple STM32F1 on my computer and manage them with openocd by using their serial id. However when i wanted to obtain their id by running st-info I discovered a weird behaviour. When I run multiple times the command, programmers disappear and only 303030303030303030303031 is left, everytime. 

Commandline output:

```
$ st-info --probe                                                                                                                                                                      
Found 2 stlink programmers                                                                                                                                                                                                                                                                
1.                                                                                                                                                                                                                                                                                        
  version:    V1J13                                                                                                                                                                                                                                                                       
  serial:     303030303030303030303031                                                                                                                                                                                                                                                    
  flash:      131072 (pagesize: 1024)                                                                                                                                                                                                                                                     
  sram:       8192                                                                                                                                                                                                                                                                        
  chipid:     0x0420                                                                                                                                                                                                                                                                      
  descr:      F1xx Value Line                                                                                                                                                                                                                                                             
2.                                                                                                                                                                                                                                                                                        
  version:    V1J13                                                                                                                                                                                                                                                                       
  serial:     49FF74067083524941522567                                                                                                                                                                                                                                                    
  flash:      131072 (pagesize: 1024)                                                                                                                                                                                                                                                     
  sram:       8192                                                                                                                                                                                                                                                                        
  chipid:     0x0420                                                                                                                                                                                                                                                                      
  descr:      F1xx Value Line                                                                                                                                                                                                                                                             

$ st-info --probe                                                                                                                                                                      
Found 1 stlink programmers                                                                                                                                                                                                                                                                
  version:    V1J13                                                                                                                                                                                                                                                                       
  serial:     303030303030303030303031                                                                                                                                                                                                                                                    
  flash:      131072 (pagesize: 1024)                                                                                                                                                                                                                                                     
  sram:       8192                                                                                                                                                                                                                                                                        
  chipid:     0x0420                                                                                                                                                                                                                                                                      
  descr:      F1xx Value Line  
```

Expected/description:

I want to have a coherent access to serial id in order to manage these devices with openocd. I am running Fedora 39 with stlink 1.7.

Thank you!

Nightwalker-87, 2023-11-24:
This toolset is not needed to manage devices with **openocd**. Both projects do not rely on each other in any case.
The STLINv1 programmer is very old and uses SCSI passthru commands over USB instead of raw USB commands.
This device type has been declared obsolete by STMicroelectronics and they do not support this programmer type any more in their tools.
In the context of the mentioned background this stlink toolset will likely maintain compatibility as is, but there will be no new features or functionalities for this device.

Nightwalker-87, 2023-11-24:
As resolving information has been given on this topic, we're closing this ticket now.

closed 2023-11-24 by Nightwalker-87