stlink-org/stlink #1362: [STM32U575RGT6]: Verification failed at offset 43008

lefebvresam, 2024-01-11:
Erasing and programming the chip with the command:

```
sudo st-flash --freq=4M  erase  0x8000000 0x100000 | sed 's/\r/\n/g'
sudo st-flash --freq=4M --debug write ./cmake_build/HMC20/develop/GCC_ARM/hmc20_test.bin 0x8000000
```

Gives as output result:

Short version:
```
sam@elitebook-845:~/git/hmc20_test$ sudo st-flash --freq=4M write ./cmake_build/HMC20/develop/GCC_ARM/hmc20_test.bin 0x8000000
st-flash 1.7.0-352-g8c581c3
2024-01-11T13:29:50 INFO common.c: STM32U575_U585: 786 KiB SRAM, 1024 KiB flash in at least 8 KiB pages.
file ./cmake_build/HMC20/develop/GCC_ARM/hmc20_test.bin md5 checksum: ad8e2a29733bb0f582e4e8c474b66c24, stlink checksum: 0x0049b610
2024-01-11T13:29:50 INFO common_flash.c: Attempting to write 47688 (0xba48) bytes to stm32 address: 134217728 (0x8000000)
2024-01-11T13:29:50 ERROR common_flash.c: Flash programming error: 0x00000080
2024-01-11T13:29:50 WARN common_flash.c: Failed to erase_flash_page(0x8000000) == -1
2024-01-11T13:29:50 ERROR common_flash.c: Failed to erase the flash prior to writing
stlink_fwrite_flash() == -1
```

Long version:
[log.txt](https://github.com/stlink-org/stlink/files/13903353/log.txt)

lefebvresam, 2024-01-11:
The last two in32's are not correctly written. If I program the same bin file with the  STM32CubeProgrammer, I can see the difference:

Flash ST-Link
[image]

Flash CubeProgrammer
[image]

lefebvresam, 2024-01-11:
Programming this:
[image]
Gives
[image]

and
[image]
Gives
[image]

lefebvresam, 2024-01-11:
>st-flash --freq=4M --debug write mem1.bin 0x800C000
[image]
[image]
---
>st-flash --freq=4M --debug write mem2.bin 0x800C000
[image]
[image]
----
>st-flash --freq=4M --debug write mem3.bin 0x800C000
[image]
[image]
----
>st-flash --freq=4M --debug write mem4.bin 0x800C000

[image]
[image]

lefebvresam, 2024-01-11:
I wrote a Python script as a workaround:

```pyhton
#!/usr/bin/python

# Python 3.10.12

import os, sys

file = ''
bytes = 0

if len(sys.argv) < 2:
    print('Give file name as first argument')
    exit(1)

file = sys.argv[1]

def append():
    try:
        with open(file, 'ab') as bf:
            print('Appending 0xFFFF to file ' + file)
            bf.write(b'\xFF\xFF\xFF\xFF')
    except OSError as e:
        print(e)
        exit(1)

def main():

    if not os.path.exists(file):
        print('File ' + file + ' does not extsis')
        exit(1)
  
    bytes = os.path.getsize(file)
    print('File size: ' + str(bytes) + 'bytes')

    if (bytes%4):
        print('File not multiple of 4 bytes')
        exit(1)
    
    ds = bytes/4

    if (ds%4 == 1):
        append()
        append()
        exit(0)

    if (ds%4 == 2):
        append()
        exit(0)

    print('Nothing to append')
    exit(0)

if __name__ == "__main__":
    main()
```

Nightwalker-87, 2024-01-15:
@lefebvresam Thanks for reporting and analysing the issue which seems to be an issue with alignment.
However I've no idea yet on how to address this appropriately in the codebase...

Ant-ON, 2024-01-16:
@Nightwalker-87 U5 chips request 16 byte alignment for write. I think need add alignment before line
[   DLOG("Starting %3u page write\n", len / sl->flash_pgsz);](https://github.com/stlink-org/stlink/blob/ae78d3639d5a0e7f15913a9f845266b29479d7bf/src/stlink-lib/flash_loader.c#L751C5-L751C5)

Alignment:
```c
    if (sl->flash_type == STM32_FLASH_TYPE_L5_U5_H5 && (len % 16))
    {
        WLOG("Data size is aligned to 16 byte");
        len += 16 - len%16;
    }
```

Nightwalker-87, 2024-01-16:
@Ant-ON I'll try. I do have a Nucleo U575ZI-Q board at hand and can try to reproduce and test.

Nightwalker-87, 2024-01-20:
$ st-flash --freq=4M --debug write mem.bin 0x8000000
```
st-flash 1.7.0-353-gae78d36-dirty
2024-01-20T12:45:55 DEBUG common.c: *** looking up stlink version ***
2024-01-20T12:45:55 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2024-01-20T12:45:55 DEBUG common.c: stlink pid     = 0x374e
2024-01-20T12:45:55 DEBUG common.c: stlink version = 0x3
2024-01-20T12:45:55 DEBUG common.c: jtag version   = 0xd
2024-01-20T12:45:55 DEBUG common.c: swim version   = 0x0
2024-01-20T12:45:55 DEBUG common.c: stlink current mode: mass
2024-01-20T12:45:55 DEBUG usb.c: JTAG/SWD freq set to 4000
2024-01-20T12:45:55 INFO usb.c: Unable to match requested speed 4000 kHz, using 3300 kHz
2024-01-20T12:45:55 DEBUG common.c: stlink current mode: mass
2024-01-20T12:45:55 DEBUG common.c: *** stlink_enter_swd_mode ***
2024-01-20T12:45:55 DEBUG common.c: Loading device parameters....
2024-01-20T12:45:55 DEBUG common.c: *** stlink_core_id ***
2024-01-20T12:45:55 DEBUG common.c: core_id = 0x0be12477
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x410fd214 at 0xe000ed00
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x20016482 at 0xe0044000
2024-01-20T12:45:55 DEBUG chipid.c: detected chip_id parameters

2024-01-20T12:45:55 DEBUG chipid.c: # Device Type: STM32U575_U585
2024-01-20T12:45:55 DEBUG chipid.c: # Reference Manual: RM0456
2024-01-20T12:45:55 DEBUG chipid.c: #
2024-01-20T12:45:55 DEBUG chipid.c: chip_id 0x482
2024-01-20T12:45:55 DEBUG chipid.c: flash_type 11
2024-01-20T12:45:55 DEBUG chipid.c: flash_size_reg 0xbfa07a0
2024-01-20T12:45:55 DEBUG chipid.c: flash_pagesize 0x2000
2024-01-20T12:45:55 DEBUG chipid.c: sram_size 0xc4800
2024-01-20T12:45:55 DEBUG chipid.c: bootrom_base 0xbf90000
2024-01-20T12:45:55 DEBUG chipid.c: bootrom_size 0x10000
2024-01-20T12:45:55 DEBUG chipid.c: option_base 0x0
2024-01-20T12:45:55 DEBUG chipid.c: option_size 0x0
2024-01-20T12:45:55 DEBUG chipid.c: flags 3

2024-01-20T12:45:55 DEBUG chipid.c: otp_base 0

2024-01-20T12:45:55 DEBUG chipid.c: otp_size 0

2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0xffff0800 at 0x0bfa07a0
2024-01-20T12:45:55 INFO common.c: STM32U575_U585: 786 KiB SRAM, 2048 KiB flash in at least 8 KiB pages.
2024-01-20T12:45:55 DEBUG common.c: *** stlink_force_debug_mode ***
2024-01-20T12:45:55 DEBUG common.c: *** stlink_status ***
2024-01-20T12:45:55 DEBUG usb.c: core status: 00030003
2024-01-20T12:45:55 DEBUG common.c:   core status: halted
file mem.bin md5 checksum: 6c2b3a8670a467e8abfaf7862ac4ff41, stlink checksum: 0x00000534
2024-01-20T12:45:55 INFO common_flash.c: Attempting to write 24 (0x18) bytes to stm32 address: 134217728 (0x8000000)
2024-01-20T12:45:55 DEBUG common.c: *** stlink_core_id ***
2024-01-20T12:45:55 DEBUG common.c: core_id = 0x0be12477
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0x000020fa to 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0xc0000000 at 0x40022028
2024-01-20T12:45:55 DEBUG usb.c: READDEBUGREG error (0x18)
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0x45670123 to 0x40022008
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0xcdef89ab to 0x40022008
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x40000000 at 0x40022028
2024-01-20T12:45:55 DEBUG common_flash.c: Successfully unlocked flash
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x40000000 at 0x40022028
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0x40000002 to 0x40022028
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x40000002 at 0x40022028
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0x40000002 to 0x40022028
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x40000002 at 0x40022028
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0x40010002 to 0x40022028
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00010000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00010000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00010000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00010000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00010000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00010000 at 0x40022020
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x40022020
[... 49 lines cut ...]
 ff ff ff ff 00 00 02 00 a5 02 00 08 00 00 00 00 08 00 02 7d 00 00 00 00
2024-01-20T12:45:55 INFO common_flash.c: Flash written and verified! jolly good!
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_debug32  0x00020000 at 0x08000004
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_reg
2024-01-20T12:45:55 DEBUG common.c: *** stlink_run ***
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_reg
2024-01-20T12:45:55 DEBUG read_write.c:  (16) ***
2024-01-20T12:45:55 DEBUG common.c: data_len = 8 0x8
 80 00 ff ff 03 00 00 f9
2024-01-20T12:45:55 DEBUG usb.c: r_idx (16) = 0xf9000003
2024-01-20T12:45:55 DEBUG common.c: *** stlink_run ***
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_read_reg
2024-01-20T12:45:55 DEBUG read_write.c:  (16) ***
2024-01-20T12:45:55 DEBUG usb.c: READREG error (0x08)
[... cut ...]
2024-01-20T12:45:55 DEBUG common.c: *** stlink_exit_debug_mode ***
2024-01-20T12:45:55 DEBUG read_write.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2024-01-20T12:45:55 DEBUG common.c: *** stlink_close ***
```

@lefebvresam I was able to reproduce your setup with a Nucleo U575ZI-Q board.
The proposed patch by @Ant-ON fixes the problem, as can be seen above.
This test used the first mem.bin as described in https://github.com/stlink-org/stlink/issues/1362#issuecomment-1887703131. I'll submit the patch to the testing branch for further verification.

closed 2024-01-20 by Nightwalker-87

lefebvresam, 2024-01-20:
I pulled the testing branch, installed again and it seems to work now. I'll no longer need my Python script. Thanks.

commit 0145baeb (2024-01-21): Fixed memory alignment for STM32L5/U5/H5 chips

(Closes #1362)