stlink-org/stlink #1363: [feature] Show all info during full erase

lefebvresam, 2024-01-22:
- [x] Programmer/board type: [V2_clone] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [Linux] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [0145baeb2e3bac31bf9d3cbd0dab38d70618d46b] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-flash] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [u575] (e.g STM32F103C8T6 (NUCLEO-F103RB))

When you use the erase function, the pages that are erased are overwritten on the screen. This is hiding a lot of info and a bit confusing. In Linux, you can use sed to convert carriage return in to line ending. But it should be better if this is resolved in code.

`st-flash --freq=4M  erase  0x8000000 0x100000 | sed 's/\r/\n/g'`

Nightwalker-87, 2024-01-23:
Bug found and fixed locally. It will be pushed to the `testing` branch soon.
However, as the Merge-Window for the upcoming Release v1.8.0 has already closed, it will not be part of this release.

lefebvresam, 2024-01-23:
It is the same during programming. The line overwrites the last line which is confusing.

Nightwalker-87, 2024-01-24:
Should be the same fix somewhere else in the code, I presume.

Nightwalker-87, 2024-02-01:
Ping @lefebvresam

lefebvresam, 2024-02-02:
Can you tell me the steps to create the right output? I tried with make debug and make install but don't see any debug information.

Nightwalker-87, 2024-02-02:
@lefebvresam Simply add the `--debug` option to the `st-flash` commands you are referring to.

lefebvresam, 2024-02-03:
Too big to paste. You can find it here:
[Uploading stlinklog.txt…]()

Nightwalker-87, 2024-02-04:
@lefebvresam I've downloaded it, so you can remove the share again, if you wish to.
It is now fixed for the `erase` command.
Did it also affect the write command?

lefebvresam, 2024-02-05:
This listing I provided is for the writing. There you see the message "writing page ..." in front of other messages.

```
Flash the chip with build/develop/source/hmc20.bin
st-flash 1.7.0-354-g32e8dcc
2024-02-04T23:56:45 INFO common.c: STM32U575_U585: 786 KiB SRAM, 1024 KiB flash in at least 8 KiB pages.
file build/develop/source/hmc20.bin md5 checksum: febabfdaa094210b9119395763fb7, stlink checksum: 0x01837946
2024-02-04T23:56:45 INFO common_flash.c: Attempting to write 250092 (0x3d0ec) bytes to stm32 address: 134217728 (0x8000000)
-> Flash page at 0x803c000 erased (size: 0x2000)
2024-02-04T23:56:45 INFO flash_loader.c: Starting Flash write for WB/G0/G4/L5/U5/H5/C0
 30/30  pages writtenARN flash_loader.c: Data size is aligned to 16 byte
2024-02-04T23:57:04 INFO common_flash.c: Starting verification of write complete
2024-02-04T23:57:06 INFO common_flash.c: Flash written and verified! jolly good!
2024-02-04T23:57:06 INFO common.c: NRST is not connected --> using software reset via AIRCR
2024-02-04T23:57:06 INFO common.c: Go to Thumb mode
```

Nightwalker-87, 2024-02-05:
Ok, thx. I'll have a look in these specific files then.

Nightwalker-87, 2024-02-06:
This is fixed by now. All related output lines are in `common_flash.c `and `flash_loader.c`.
All carriage return `\r` characters found have been replaced by newline `\n` characters.
The ticket will be closed by automation upon reference in one of the next commits.

lefebvresam, 2024-02-06:
How can I test it?

Nightwalker-87, 2024-02-06:
Well you can replace the characters in the mentioned files yourself locally and recompile. :wink:
I'll likely not generate a separate commit solely for it.
It will be pushed bundled with other fixes instead, as soon as these are resolved as well.

commit 2ca9f6aa (2024-02-16): Post release patch for v1.8.0

- Fixed terminal screen output (Closes #1363)
- Fixed compatibility with gcc-14 (Closes #1365)
- Updated contact info in CODE_OF_CONDUCT
- Updated man files & related install path
- Updated libusb to v1.0.27 for Windows builds
- Updated cmake_policy to CM0153 NEW
commit 6a6718b3 (2024-02-16): Post release patch for v1.8.0

- Fixed terminal screen output (Closes #1363)
- Fixed compatibility with gcc-14 (Closes #1365)
- Updated contact info in CODE_OF_CONDUCT
- Updated man files & related install path
- Updated libusb to v1.0.27 for Windows builds
- Updated cmake_policy to CM0153 NEW
closed 2024-02-16 by Nightwalker-87