stlink-org/stlink #1365: Can't build stlink 1.8.0 for Fedora

Vascom, 2024-02-02:
- [ ] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [ ] Programmer/board type: [no]
- [ ] Operating system an version: [Fedora GNU/Linux Rawhide]
- [ ] **stlink tools version** and/or git commit hash: [1.8.0] 
- [ ] stlink commandline tool name: [no]
- [ ] Target chip (and board, if applicable): [no]

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:
```
/builddir/build/BUILD/stlink-1.8.0/src/stlink-lib/chipid.c: In function ‘process_chipfile’:
/builddir/build/BUILD/stlink-1.8.0/src/stlink-lib/chipid.c:68:22: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
   68 |   ts = calloc(sizeof(struct stlink_chipid_params), 1);
      |                      ^~~~~~
/builddir/build/BUILD/stlink-1.8.0/src/stlink-lib/chipid.c:68:22: note: earlier argument should specify number of elements, later size of each element
/builddir/build/BUILD/stlink-1.8.0/src/stlink-lib/chipid.c: In function ‘process_chipfile’:
/builddir/build/BUILD/stlink-1.8.0/src/stlink-lib/chipid.c:68:22: error: ‘calloc’ sizes specified with ‘sizeof’ in the earlier argument and not in the later argument [-Werror=calloc-transposed-args]
   68 |   ts = calloc(sizeof(struct stlink_chipid_params), 1);
      |                      ^~~~~~
/builddir/build/BUILD/stlink-1.8.0/src/stlink-lib/chipid.c:68:22: note: earlier argument should specify number of elements, later size of each element
```

Expected:
Successful build.

I can't build stlink 1.8.0 for Fedora. Please help me fix it?

Nightwalker-87, 2024-02-02:
Interesting, it does not occur on debian (testing).
Please share the build-tool versions of your current toolchain to allow for further investigation.

Vascom, 2024-02-02:
Versions https://kojipkgs.fedoraproject.org//work/tasks/3120/112823120/root.log
Full build log https://kojipkgs.fedoraproject.org//work/tasks/3120/112823120/build.log

Nightwalker-87, 2024-02-02:
eh, could you strip that down to the relevant information on the build tools used in the form of name, version, etc. ?

Vascom, 2024-02-03:
gcc 14.0.1
cmake 3.27.7
gtk3 3.24.41
libusb1 1.0.26
systemd 255.3

Nightwalker-87, 2024-02-04:
@Vascom It's a gcc14 issue: 
- https://www.mail-archive.com/elfutils-devel@sourceware.org/msg06687.html 
- https://forums.fedoraforum.org/showthread.php?331882-Latest-gcc-compiler-anomalie&p=1879224
-  https://lore.kernel.org/all/20240202013221.365561-1-raj.khem@gmail.com/T/

Try to use gcc13.x for now. (which works, I used it for the compilation and package generation.)
As far as I can see it does not seem to be a toolset related issue.
However we would need to address it somewhen...

Nightwalker-87, 2024-02-04:
@Vascom Can you do some research on the web in order to point to the necessary fixes in the codebase?
It would be desirable if there were some more contributors to assist as debugging itself is very time consuming.

Nightwalker-87, 2024-02-04:
@Vascom: I seem to have found an unexpected solution for this issue: Try to swap the arguments of `calloc` - it should resolve the issue. There are no further similar findings by the `gcc-14` compiler.
Please verify this at your side and report back.
Unless there is not more to it, I'll push the fix soon together with other resolved issues.

Vascom, 2024-02-06:
Thanks.
With gcc 13.2.1 there is no problem. But upcoming Fedora 40 have gcc 14.

Vascom, 2024-02-06:
Also if I swap calloc arguments in src/stlink-lib/chipid.c line 68 it compiled succesfully with gcc 13 and gcc 14.

commit 2ca9f6aa (2024-02-16): Post release patch for v1.8.0

- Fixed terminal screen output (Closes #1363)
- Fixed compatibility with gcc-14 (Closes #1365)
- Updated contact info in CODE_OF_CONDUCT
- Updated man files & related install path
- Updated libusb to v1.0.27 for Windows builds
- Updated cmake_policy to CM0153 NEW
commit 6a6718b3 (2024-02-16): Post release patch for v1.8.0

- Fixed terminal screen output (Closes #1363)
- Fixed compatibility with gcc-14 (Closes #1365)
- Updated contact info in CODE_OF_CONDUCT
- Updated man files & related install path
- Updated libusb to v1.0.27 for Windows builds
- Updated cmake_policy to CM0153 NEW
closed 2024-02-16 by Nightwalker-87