stlink-org/stlink #1366: [feature] Chaching and diffing bin/hex

lefebvresam, 2024-02-05:
In most of the cases when you are programming you only change small pieces of code. It means that the bin/hex is changed only  slightly based on the previous one. When you program the device, all blocks are programmed again. It means that even when you change a small string value, all 40 or 50 memory blocks are re-programmed again and you have to wait all the times for 40sec because the max programming speed is 4Mbps. Should it be possible to check which blocks have been changed based on the previous bin/hex and only program those which are different? It could speedup the development cycle a lot. It could be an option to provide --non-caching or something like that to reprogram the whole flash.

Nightwalker-87, 2024-02-05:
Likely a duplicate of #1238.

Nightwalker-87, 2024-08-02:
Adressed in #1397.

closed 2024-08-02 by Nightwalker-87