stlink-org/stlink #1376: Debian 11 x64 doesn't work with v1.8.0 because of incompatible glibc

xor-gate, 2024-03-06:
Because for some reason the issue tracker is closed for public collaboration I got this on my blog e-mail address:

> Hi @xor-gate !
>
> I wanted to report this as a Github issue but issues are disabled for
noncontributors.
> 
> Anyway, Debian 11 is listed on the stlink-org/stlink as a supported os,
> but the latest (v1.8.0) release fails to run on Debian 11 because it uses
> a too-new glibc:

> st-flash: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found
> (required by st-flash)

>  verified that my glibc is updated. The current version for Debian 11 is
> 2.31 (2.31-13+deb11u8).

>  thought you might want to know about this as the README is incorrect
> now. Debian 12 uses glibc 2.36, which is new enough to work.

> Best Regards
> [@polprog](https://github.com/polprog)

Nightwalker-87, 2024-03-08:
It is currently unknown from where this dependency derived - it has not been introduced deliberately and as to my current knowledge, the build system does not require any specific minimum version here either.

Nightwalker-87, 2024-03-08:
@polprog: I'd recommend to use v1.7.0 via the Debian backports package source for now on your system.

This are the only dependencies we have when compiling the Debian package:
`set(CPACK_DEBIAN_PACKAGE_DEPENDS "pkg-config, build-essential, debhelper (>=9), cmake (>= 3.13.0), libusb-1.0-0-dev (>= 1.0.22)")`

Nightwalker-87, 2024-04-06:
@polprog: I have had another close look at this:
As Debian 11 (Bullseye) is moving to the _oldoldstable_ state on 30 Jun 2024 to enter it's LTS period, it appears not worth the effort to reconfigure the used development system to explicitly work with an older build environment (or switch around back and forth continuously, which is another potential source for errors), while most other distributions update toolchains even more frequently than _testing_ release state does. There are very likely not many development systems around working with such an relatively old package source (as this appears to be in contrast to new development efforts targeting recent systems on the installed base). Thus I've decided to end support for Debian 11 this summer at the latest.

commit ece34efb (2024-10-03): Updated info on OS support. (Closes #1376)
closed 2024-04-06 by Nightwalker-87