stlink-org/stlink #1379: [feature] Just a question .. about next release and ... clones ...

dikdom, 2024-03-11:
Hello,

Is there a release date for 1.8.1?
Is the newer st-tools recognize properly my STM32F401? (0x0423 - F4xx (low power) )

Do you happen to plan to add more (or just some) clone device IDs? What I have in my hand is a black-pill AT32F403A (and making possible to create custom chip? I don't know how hard/stupid that would be..)

```
...stlink-1.7.0\bin>st-info.exe --probe
Found 1 stlink programmers
  version:    V2J43S7
  serial:     710040000800004150514E4E
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x0347

```
---

**NOTE: In order to offer sufficient and the best possible support, please read /CONTRIBUTING.md and follow the given instructions _before_ submitting a ticket.**

**Bug reports and/or feature requests will be deleted, if they violate our contribution guidelines and if no issue-template is used!** Thank you for your support.

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific request.

- [x] Programmer/board type: [STLINK V2] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: [win10] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [1.7.0] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-info] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [enter here] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

Commandline output:

```
OUTPUT/ERROR of the commandline tool(s)
```

Expected/description:

`short description of the expected value`

Nightwalker-87, 2024-03-12:
No, there is no release date for any upcoming release. I don't expect it to happen very soon either, as we have just seen a new major release. We no longer support any clone devices. This topic as well as the reasons for this decision have already been covered intensively in this project.

Nightwalker-87, 2024-03-12:
> Is the newer st-link tools recognize properly my STM32F401? (0x0423 - F4xx (low power) )

As long as it is an original part and correctly detected and handled by the official ST-Link Tools from ST Microelectronics, one may expect it to work with this toolset as well.

Please ensure to use the latest release whenever possible.

As all points have been addressed, I'm closing this issue now as resolved.