stlink-org/stlink #1384: STM32 Nucleo-H563ZI Board not recognized

Nightwalker-87, 2024-04-06:
### Discussed in https://github.com/stlink-org/stlink/discussions/1381

<div type='discussions-op-text'>

<sup>Originally posted by **DF4OR** March 26, 2024</sup>
Hi.
Beginner here. 
I have sucessfully used other boards (e.g. Nucleo-F091RC) with the current st-link utils on Linux. The Nucleo-H563ZI board in contrast is not properly recognized and cannot be programmed.

Output of lsusb 
`Bus 003 Device 017: ID 0483:374e STMicroelectronics STLINK-V3`

Output of st-info --probe
```Found 1 stlink programmers
  version:    V3J10
  serial:     003A001B3132511138363431
  flash:      0 (pagesize: 0)
  sram:       0
  chipid:     0x000
  dev-type:   unknown
```
Outout of st-info --version
`v1.8.0`

System
* Arch Linux, kernel 6.8.1-arch1-1
* stlink utils installed with pacman

Any idea what's going on?

TIA,
Ekki
</div>

muxelplexer, 2024-05-14:
I'd be very much interested in helping on this issue as i've been trying to setup my Nucleo-H563ZI through stlink myself. Would you have any clues/guidance what is missing for this board for someone who hasn't been too involved with the codebase yet? @Nightwalker-87

As far as i've seen it should only be the flash that is not being picked up correctly? Or is it actually the Stlink-3VPWR that is throwing it off?

Nightwalker-87, 2024-05-14:
@muxelplexer Thanks for the feedback. However I'm not really sure, if I can give any useful advice here, as I am not aware of the technical specifics of this board and how they relate to certain parts in the code base. One should know that there are coexisting older and newer parts of the st-link library, which derive from different ways of implementation. This has not fully been resolved yet and I don't know if/how this affects this issue in any kind.

antoinh, 2024-06-26:
I'm also interested on getting this to work with STM32H5 series. One of the quick thing I notice is that the stm32H5 require connection on access port 1 instead of the traditional port 0 when using SWD. There might be some other changes but that is certainly a point to start with these new chips.

In ST-LINK utility  or STCubeProgramme there is such an option but I don't see this option in the open source stlink.

See this thread from ST where the ST employee mentions the connection via port1. I can also see that stm32cube programmer does the switch automatically to port1:

https://community.st.com/t5/stm32cubeide-mcus/unable-to-flash-nucleo-h563zi-with-stm32cubeide-v1-14/m-p/618423/highlight/true#M22695

[image]

ypearson-bdai, 2024-07-01:
Any news here?

tonyh1988, 2024-07-02:
No, sorry I don't have time to investigate this further at the moment. I just thought I would comment my findings to help someone that might want to dive deeper into this eventually or maybe even myself if I want to dive deeper on this later

tonyh1988, 2024-07-02:
That being said the stm32cube programmer command line interface is freely available and should in theory work with STM32H5. I will need to investigate this further until the open source stlink driver catch up with these newer MCUs.

Nightwalker-87, 2024-10-03:
We are still waiting for any constructive input here...

ypearson-bdai, 2024-10-04:
> We are still waiting for any constructive input here...

I used this and seems to work Ok:

https://github.com/patrickelectric/platform-ststm32/tree/h5

Nightwalker-87, 2024-10-11:
Yes, well, not sure how this can help us along with our codebase...

c-grant, 2024-12-14:
Just a thought here. When I had an H7 board I had to add all the Chip ID/Memory map stuff for the H7 processors. I did a quick look in the code and I didn't see anything in gdb-server.c for the H5.

Nightwalker-87, 2025-01-06:
As to my knowledge `gdb-server.c` should not be needed for chip or board recognition.

Nightwalker-87, 2025-01-08:
I now have this hardware available for testing.However, I need detailed instructions on how to reproduce this on linux, preferably Debian.

Nightwalker-87, 2025-01-09:
Problem report from @DF4OR verified on Debian 6.12.6-1 (2024-12-21).

Nightwalker-87, 2025-01-11:
I found this [reference](https://community.st.com/t5/stm32-mcus-products/nucleo-stm32h5-debugging-openocd/td-p/618225) for the implementation within the openocd project, which my help resolving this issue here as well. Implementation obviously needs to go to /src/stlink-lib/usb.c. further ideas are welcome.

Nightwalker-87, 2025-01-11:
@Ant-ON: You contributed to this file earlier regarding hardware specific commands.
Any idea on where to and how to define the access port number?

tonyh1988, 2025-01-11:
In your case I guess you could look at openocd documentation and try to find the correct commands to use a different access port.

Nightwalker-87, 2025-01-11:
@tonyh1988: I'm not familiar with this part of the codebase either, that's why I asked for support with focus on previous authors.

tonyh1988, 2025-01-11:
I understand, when I initially looked quickly into this this summer I concluded it was not a super quick fix. One would need to dive deeper and understand what the difference is with official st cube programmer. Since I was able to use the official st tools to program my device I moved on with my project and didn't look any further into this.

 FYI @antoinh is a different account of mine

lbthomsen, 2025-12-12:
STM32H562 using CubeProgrammer:

```
  11:38:51:811 : UR connection mode is defined with the HWrst reset mode
  11:38:51:997 : ST-LINK SN  : 066AFF3632504E3043064426
  11:38:51:997 : ST-LINK FW  : V2J46M33
  11:38:51:997 : Board       : --
  11:38:51:997 : Voltage     : 3.24V
  11:38:51:997 : Warning:  Connection to AP 0 requested and failed, Connection established with AP 1
  11:38:51:998 : SWD freq    : 4000 KHz
  11:38:51:998 : Connect mode: Normal
  11:38:51:998 : Reset mode  : Software reset
  11:38:51:998 : Device ID   : 0x484
  11:38:51:998 : Revision ID : Rev X
  11:38:51:998 : halt ap 1 
  11:38:51:998 : r ap 1 reg 15 PC   0x08000D18
  11:38:51:999 : halt ap 1 
  11:38:51:999 : r ap 1 reg 15 PC   0x08000D18
  11:38:52:000 : Database: Config 16 is active.
  11:38:52:000 : flash loader /home/lth/st/STM32Cube/STM32CubeProgrammer/bin/FlashLoader/0x484.stldr is loaded
  11:38:52:000 : Debug in Low Power mode enabled.
  11:38:52:000 : halt ap 1 
  11:38:52:000 : r ap 1 reg 15 PC   0x08000D18
  11:38:52:001 : SFSP Version: v2.6.0
  11:38:52:001 : halt ap 1 
  11:38:52:001 : r ap 1 reg 15 PC   0x08000D18
  11:38:52:175 : UPLOADING OPTION BYTES DATA ...
  11:38:52:175 :   Bank          : 0x00
  11:38:52:175 :   Address       : 0x40022050
  11:38:52:175 :   Size          : 112 Bytes
  11:38:52:175 :   Bank          : 0x01
  11:38:52:175 :   Address       : 0x40022070
  11:38:52:175 :   Size          : 16 Bytes
  11:38:52:175 :   Bank          : 0x02
  11:38:52:175 :   Address       : 0x40022080
  11:38:52:175 :   Size          : 16 Bytes
  11:38:52:175 :   Bank          : 0x03
  11:38:52:175 :   Address       : 0x400220e0
  11:38:52:175 :   Size          : 16 Bytes
  11:38:52:176 :   Bank          : 0x04
  11:38:52:176 :   Address       : 0x400221e0
  11:38:52:176 :   Size          : 16 Bytes
  11:38:52:176 :   Bank          : 0x05
  11:38:52:176 :   Address       : 0x40022090
  11:38:52:176 :   Size          : 8 Bytes
  11:38:52:176 :   Bank          : 0x06
  11:38:52:176 :   Address       : 0x400220f0
  11:38:52:176 :   Size          : 8 Bytes
  11:38:52:177 :   Bank          : 0x07
  11:38:52:177 :   Address       : 0x400221f0
  11:38:52:177 :   Size          : 8 Bytes
  11:38:52:177 :   Bank          : 0x08
  11:38:52:177 :   Address       : 0x400220f8
  11:38:52:177 :   Size          : 8 Bytes
  11:38:52:177 :   Bank          : 0x09
  11:38:52:177 :   Address       : 0x400221f8
  11:38:52:177 :   Size          : 8 Bytes
  11:38:52:177 : UPLOADING OPTION BYTES DATA ...
  11:38:52:177 :   Bank          : 0x00
  11:38:52:177 :   Address       : 0x40022050
  11:38:52:177 :   Size          : 112 Bytes
  11:38:52:177 :   Bank          : 0x01
  11:38:52:177 :   Address       : 0x40022070
  11:38:52:177 :   Size          : 16 Bytes
  11:38:52:177 :   Bank          : 0x02
  11:38:52:177 :   Address       : 0x40022080
  11:38:52:177 :   Size          : 16 Bytes
  11:38:52:178 :   Bank          : 0x03
  11:38:52:178 :   Address       : 0x400220e0
  11:38:52:178 :   Size          : 16 Bytes
  11:38:52:178 :   Bank          : 0x04
  11:38:52:178 :   Address       : 0x400221e0
  11:38:52:178 :   Size          : 16 Bytes
  11:38:52:178 :   Bank          : 0x05
  11:38:52:178 :   Address       : 0x40022090
  11:38:52:178 :   Size          : 8 Bytes
  11:38:52:178 :   Bank          : 0x06
  11:38:52:178 :   Address       : 0x400220f0
  11:38:52:178 :   Size          : 8 Bytes
  11:38:52:178 :   Bank          : 0x07
  11:38:52:178 :   Address       : 0x400221f0
  11:38:52:178 :   Size          : 8 Bytes
  11:38:52:178 :   Bank          : 0x08
  11:38:52:178 :   Address       : 0x400220f8
  11:38:52:178 :   Size          : 8 Bytes
  11:38:52:179 :   Bank          : 0x09
  11:38:52:179 :   Address       : 0x400221f8
  11:38:52:179 :   Size          : 8 Bytes
  11:38:52:179 : UPLOADING ...
  11:38:52:179 :   Size          : 1024 Bytes
  11:38:52:179 :   Address       : 0x8000000
  11:38:52:179 : Read progress:
  11:38:52:179 : halt ap 1 
  11:38:52:179 : r ap 1 reg 15 PC   0x08000D18
  11:38:52:180 : Data read successfully
  11:38:52:180 : Time elapsed during the read operation is: 00:00:00.017
```

Notice the "ap 1" (presumably Access Port 1).  The Programmer UI switches the Access Port to 1 automatically.