stlink-org/stlink #1391: [STM32 device name]: $YourTitle

AJ528, 2024-05-05:
Note: this could be related to issue #658 

---

- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: STLINK V3
- [x] Operating system and version: Linux Mint 21.3
- [x] **stlink tools version** and/or git commit hash: v1.8.0
- [x] stlink commandline tool name: st-util
- [x] Target chip (and board, if applicable): STM32WL55 (NUCLEO-WL55JC1)

Further we kindly ask you to describe the detected problem as detailed as possible and to add debug output if available, by using the following template:

I am using st-util as a GDB server to remote debug my STM32 Nucleo board. I have 2 versions of st-util. One I downloaded from the package manager via `sudo apt get install stlink` and the other I compiled from source. Both are version 1.8.0.

When I use the package manager version of st-util to debug code, it takes a fairly long time to single-step through code. Somewhere on the order of 7-10 seconds per step.

I downloaded the source code, compiled it using `make release`, and ran the st-util command from the Release folder. When using that version, I can step through the code much quicker, something like 1-2 seconds per step.

Both versions of st-util report v1.8.0, but I did notice that the compiled version was looking for chipids in `/usr/local/share/stlink/config/chips` whereas the package manager version had put them in `/usr/local/share/stlink/chips`. So apparently there is at least 1 difference between the two versions.

I would expect both versions of the st-util to act identically and be able to step through the code quickly. I almost gave up on st-util because 7-10 seconds per step is way too long.

Nightwalker-87, 2024-05-08:
We have no influence on the package configuration downstream including repacking in the distro repository.
Thus there indeed may be changes, as described, between different distros and to this projects repo.

Nightwalker-87, 2024-05-08:
In this context I recommend to use the release-version distributed by this project.

closed 2024-05-08 by Nightwalker-87