stlink-org/stlink #1402: [STM32 F407VET6]: WARN usb.c: skipping ST device: 0x483:0x5720

amuerta, 2024-06-11:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

-  Programmer/board type: [STM32 F407VET6] 
- Operating system an version: [NixOS 23.11 and 24.05, also Debain 12] 
-  **stlink tools version** and/or git commit hash: [v1.8.0 and 32ce4bf ] (both)
-  stlink commandline tool name: [st-info and all other] 
-  Target chip : [STM32 F407VET6] 

Details:
i got a an assignment to create a simple project for any of STM32 family boards. Professor have not explained anything about how exactly we should do the project by calling it out as "its your problem".

Since good results are important for me i bought STM32F407VET6 board to try all i can.
Only reasonable tool i found for the job was this project. I encountered a problem like in #1274 , but changing configuration file did not help. 

I tried running stlink on both NixOS and Debian, since i expect having lack of knowledge about first, so trying to run it on different Operating System might resolve and issue.
No. Its the same on both.

I tried building the project from source and changing [49-stlinkv3.rules](https://github.com/stlink-org/stlink/blob/develop/config/udev/rules.d/49-stlinkv3.rules) file to target different versions (1,2,3) of STLINK, also specified 5720 as product id.
Didn't work either.

I have no experience developing in embedded environment and don't even know if its a bug or just my board is not supported.
Ill be very thankful for any help.

Command line output:

```
st-info --probe 
  2024-06-11T19:14:42 WARN usb.c: skipping ST device : 0x483:0x5720)
  Found 0 stlink programmers
  
st-info --descr
  libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/001/010, errno=13
  libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
```

GerryFerdinandus, 2024-06-13:
@amuerta 
0x483:0x5720 is for ST with mass storage
https://the-sz.com/products/usbid/index.php?v=0x0483

What you can try is to remove this mass storage mode through CubeProgrammer.
https://community.st.com/t5/stm32-mcus-products/question-can-i-disable-the-nucleo-onboard-stlink-mass-storage/td-p/464705

amuerta, 2024-07-29:
Following the instructions helped, now everything works as intended, thanks!:)