stlink-org/stlink #1407: [NUCLEO-H563ZI]: Cannot program board

ypearson-bdai, 2024-07-01:
- [ /V3] Programmer/board type: [enter here] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [Ubuntu ] Operating system an version: [enter here] (e.g Linux, Windows)
- [1.8.0 ] **stlink tools version** and/or git commit hash: [enter here] (e.g v1.6.1/git-d0416149)
- [ st-flash --connect-under-reset write build/NUCLEO-H563ZI.bin 0x8000000] stlink commandline tool name: [enter here] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [ STM32H56x/573] Target chip (and board, if applicable): [enter here] (e.g STM32F103C8T6 (NUCLEO-F103RB))

`st-flash --connect-under-reset write build/NUCLEO-H563ZI.bin 0x8000000`
`st-flash 1.8.0`
`2024-07-01T11:46:40 WARN common.c: NRST is not connected`
`2024-07-01T11:46:40 ERROR common.c: Soft reset failed: error write to AIRCR`
`2024-07-01T11:46:40 ERROR common.c: Can not connect to target. Please use 'connect under reset' and try again`
`Failed to connect to target`
`Failed to parse flash type or unrecognized flash type`

ashiqguntupalli, 2024-07-31:
Any update on this issue?
I do have the same problem while programming the NUCLEO-H563ZI 144 pin board.

Nightwalker-87, 2024-08-11:
Duplicate of #1384.