stlink-org/stlink #1410: st-flash.exe - Application Error (0xc000007b)

arthurrGama, 2024-07-05:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue

In order to allow developers to isolate and target your respective issue, please take some time to select the check boxes below and fill out each of the following items appropriate to your specific problem.

- [x] Programmer/board type: ST-Link/V2 (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x] Operating system an version: Win10 Pro 22H2 x64 bits (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: stlink v1.8.0 / https://github.com/stlink-org/stlink/releases/tag/v1.8.0 (e.g v1.6.1/git-d0416149)

I've been using the STLink tool on my main machine for a few days now, and it works perfectly. However, I'm now trying to use it on another machine, which is actually a super tablet running Windows :) and using an Intel Atom x64 processor.

When I try to run the code, it complains about libusb-1.0.dll. I installed it and placed it in the executable directory, but when I try to run it, I get the following error:

_**st-flash.exe - Application Error**
**The application was unable to start correctly (0xc000007b). Click OK to close the application.**_

I'm unable to make this work. I've installed the ST-Link drivers, but I feel it might have something to do with x32 --> x64 compatibility. I haven't found the solution yet. It's exactly the same as my main machine, which works perfectly.

Nightwalker-87, 2024-07-17:
There is no known general incompatibility with Windows 10, neither for the x86 nor for the x86-64 variant.
Likely linked to #1364. Please stay with version v1.7.0 in the meanwhile.

Nightwalker-87, 2024-08-11:
This issue is handled in #1364.

commit e4931097 (2024-08-11): General Project Update

- Fixed compatibility with gcc-14 [-Werror=calloc-transposed-args]
- Updated compiling instructions and build script for Windows (Closes #1364) (Closes #1410)
- Updated CHANGELOG.md
- Minor formatting fixes
closed 2024-08-11 by Nightwalker-87