stlink-org/stlink #1425: [STM32F042]:  Unable to write option bytes

kajusK, 2024-09-13:
- [x] I made serious effort to avoid creating duplicate or nearly similar issue
- [x ] Programmer/board type: [STLINK V3 Mini] (e.g STLINK /V1, /V2, /V2-onboard, /V2-clone, /V3)
- [x ] Operating system an version: [Linux] (e.g Linux, Windows)
- [x] **stlink tools version** and/or git commit hash: [v1.8.0] (e.g v1.6.1/git-d0416149)
- [x] stlink commandline tool name: [st-flash] (e.g `st-info`, `st-flash`, `st-trace`, `st-util`)
- [x] Target chip (and board, if applicable): [STM32F042G6] (e.g STM32F103C8T6 (NUCLEO-F103RB))

Commandline output:

```
st-flash --debug --area=option write 0x007755aa
st-flash 1.8.0
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/031, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
2024-09-13T13:54:52 DEBUG common.c: *** looking up stlink version ***
2024-09-13T13:54:52 DEBUG common.c: st vid         = 0x0483 (expect 0x0483)
2024-09-13T13:54:52 DEBUG common.c: stlink pid     = 0x374e
2024-09-13T13:54:52 DEBUG common.c: stlink version = 0x3
2024-09-13T13:54:52 DEBUG common.c: jtag version   = 0x7
2024-09-13T13:54:52 DEBUG common.c: swim version   = 0x0
2024-09-13T13:54:52 DEBUG common.c: stlink current mode: mass
2024-09-13T13:54:52 DEBUG usb.c: JTAG/SWD freq set to 0
2024-09-13T13:54:52 DEBUG common.c: stlink current mode: mass
2024-09-13T13:54:52 DEBUG common.c: *** stlink_enter_swd_mode ***
2024-09-13T13:54:52 DEBUG common.c: Loading device parameters....
2024-09-13T13:54:52 DEBUG common.c: *** stlink_core_id ***
2024-09-13T13:54:52 DEBUG common.c: core_id = 0x0bb11477
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0x410cc200 at 0xe000ed00
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0x10006445 at 0x40015800
2024-09-13T13:54:52 DEBUG chipid.c: detected chip_id parameters

2024-09-13T13:54:52 DEBUG chipid.c: # Device Type: STM32F04x
2024-09-13T13:54:52 DEBUG chipid.c: # Reference Manual: RM0091
2024-09-13T13:54:52 DEBUG chipid.c: #
2024-09-13T13:54:52 DEBUG chipid.c: chip_id 0x445
2024-09-13T13:54:52 DEBUG chipid.c: flash_type 2
2024-09-13T13:54:52 DEBUG chipid.c: flash_size_reg 0x1ffff7cc
2024-09-13T13:54:52 DEBUG chipid.c: flash_pagesize 0x400
2024-09-13T13:54:52 DEBUG chipid.c: sram_size 0x1800
2024-09-13T13:54:52 DEBUG chipid.c: bootrom_base 0x1fffec00
2024-09-13T13:54:52 DEBUG chipid.c: bootrom_size 0xc00
2024-09-13T13:54:52 DEBUG chipid.c: option_base 0x1ffff800
2024-09-13T13:54:52 DEBUG chipid.c: option_size 0x10
2024-09-13T13:54:52 DEBUG chipid.c: flags 0

2024-09-13T13:54:52 DEBUG chipid.c: otp_base 0

2024-09-13T13:54:52 DEBUG chipid.c: otp_size 0

2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0xffff0020 at 0x1ffff7cc
2024-09-13T13:54:52 INFO common.c: STM32F04x: 6 KiB SRAM, 32 KiB flash in at least 1 KiB pages.
2024-09-13T13:54:52 DEBUG common.c: *** stlink_force_debug_mode ***
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0xe0042004
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0x00000300 to 0xe0042004
2024-09-13T13:54:52 DEBUG common.c: *** stlink_status ***
2024-09-13T13:54:52 DEBUG usb.c: core status: 01030003
2024-09-13T13:54:52 DEBUG common.c:   core status: halted
2024-09-13T13:54:52 WARN option_bytes.c: About to write option byte   0x7755aa to 0x1ffff800.
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x4002200c
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0x00000080 at 0x40022010
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0x45670123 to 0x40022004
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0xcdef89ab to 0x40022004
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x40022010
2024-09-13T13:54:52 DEBUG common_flash.c: Successfully unlocked flash
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x40022010
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0x45670123 to 0x40022008
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0xcdef89ab to 0x40022008
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0x00000200 at 0x40022010
2024-09-13T13:54:52 DEBUG common_flash.c: Successfully unlocked flash option
2024-09-13T13:54:52 WARN option_bytes.c: Only full write of option bytes area is supported
2024-09-13T13:54:52 ERROR option_bytes.c: Flash option write failed!
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0x00000200 at 0x40022010
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0000000000 to 0x40022010
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_read_debug32  0000000000 at 0x40022010
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0x00000080 to 0x40022010
stlink_write_option_bytes32() == -1
2024-09-13T13:54:52 DEBUG common.c: *** stlink_exit_debug_mode ***
2024-09-13T13:54:52 DEBUG read_write.c: *** stlink_write_debug32 0xa05f0000 to 0xe000edf0
2024-09-13T13:54:52 DEBUG common.c: *** stlink_close ***
```

Expected/description:
The write of option bytes always fails with `Only full write of option bytes area is supported`. I'm providing all 4 bytes in the command. Why is that write not accepted?

Nightwalker-87, 2024-10-03:
@kajusK: Please ensure you have write access on your interface (group `dialout`) and that the device hardware is recognised correctly:

```
libusb: error [get_usbfs_fd] libusb couldn't open USB device /dev/bus/usb/003/031, errno=13
libusb: error [get_usbfs_fd] libusb requires write access to USB device nodes
```

Nightwalker-87, 2024-10-19:
Ping @kajusK. If there is no further response, this ticket will be closed in 7 days.

kajusK, 2024-10-23:
@Nightwalker-87 Oops, sorry, I've missed the previous response. I don't have the hardware with me currently, I'll give it a try over the weekend.

I thought the hw shall be recognized and accessible when it's possible to use it to flash a firmware into the device...

Nightwalker-87, 2024-11-09:
Support for writing option bytes on STM32F042 is present at least since #1181.
The issue seems to be caused by missing write permissions on the host system.
Please ensure correct settings and take care of it independently.
This is obviously not an issue cause by the toolset itself.

closed 2024-11-09 by Nightwalker-87